siemens/meta-iot2000 #7: Node.js Version

kahveci, 2017-01-30:
The latest example image (v2.1.2) has Node.js version 4.2.4. Could you please upgrade it to the latest stable version 6.9.4 in the next SD-Card example image? Thanks.

SaschaWeisenberger, 2017-02-01:
Yes we plan to use the LTS version (I think we use 6.9.2 because there is already an official recipe for it) at least in the github repo. Chances are good that it will find its way into the binary image in too ;)

kahveci, 2017-02-01:
Thanks for the reply Sascha. Do you have a road-map for the example OS image updates?

SaschaWeisenberger, 2017-02-01:
Not officially ;) it partly depends on people like you who report issues and requirements. So if you have any demands or ideas please let us know!

teozfrank, 2017-02-14:
It would be ideal if we could specify what version of node.js, I for example need at minimum version 4.7.0 but the example image only has 4.2.0, Regards Frank.

SaschaWeisenberger, 2017-04-13:
Updated to 6.9.2 now.

closed 2017-04-13 by SaschaWeisenberger