siemens/meta-iot2000 #9: Oracle JDK / JRE

teozfrank, 2017-02-14:
As the title suggests, I cannot find anywhere how to install oracle java onto the IOT 2040, it would be nice if the image had it already included or to select what version.

Regards,
Frank.

jan-kiszka, 2017-02-14:
I strongly suspect that binary-only Oracle Java will not work: The Intel Quark requires applications to be built with special compiler options to work around processor errata. That's why we build everything in Yocto and cannot use a standard distribution.

If the 32-bit non-rpm Linux version you can download from Oracle does not run as-is, you need to use the OpenSDK/JRE that is part of the example image.

teozfrank, 2017-02-15:
What about this? http://git.yoctoproject.org/cgit/cgit.cgi/meta-oracle-java/

jan-kiszka, 2017-02-15:
That's just an installer recipe for the binary package. Nothing is built from sources here either.

teozfrank, 2017-02-16:
> OpenSDK/JRE that is part of the example image.

How do I use this?

jan-kiszka, 2017-02-16:
Oh, it isn't in fact. It's only used during the build. But you can do this by adding IMAGE_INSTALL_append=" openjre-8" in your local.conf, e.g.

jan-kiszka, 2017-04-10:
Please reopen if there are further questions.

closed 2017-04-10 by jan-kiszka

jan-kiszka, 2017-05-10:
FWIW, I just had to validate Oracle Java 8 (update 131) on the latest image, and it apparently works fine. Just like OpenJRE, built from sources via Yocto as I described above.

teozfrank, 2017-06-02:
does that work with an existing build or does that have to be done from a custom build? If that is not the case where is the local.conf located to add that?

jan-kiszka, 2017-06-02:
We didn't try with current official release but went directly to the development version that is generated from master.