siemens/meta-iot2000 #13: When trying to bitbake , meta-intel/conf/layer.conf not found

jancumps, 2017-04-09:
I'm using the instructions in the readme for building meta-iot2000
https://github.com/siemens/meta-iot2000/tree/master/meta-iot2000-bsp

prereqs installed, Prepare, Download, enter build steps all work.
Config done. both conf scripts adapted as indicated.

When bitbaking, I get the next error immediately:

> ERROR: Unable to parse /home/build/poky/meta-intel/conf/layer.conf: [Errno 2] file /home/build/poky/meta-intel/conf/layer.conf not found

It seems to think that my build folder is /home/build/ , although I initiated everything from /home/jancumps/iot2020/ .

```
jancumps@jancumps-VirtualBox:~/iot2020/iot2000-build$ bitbake core-image-minimalERROR: Traceback (most recent call last):
  File "/home/jancumps/iot2020/poky/bitbake/lib/bb/cookerdata.py", line 179, in wrapped
    return func(fn, *args)
  File "/home/jancumps/iot2020/poky/bitbake/lib/bb/cookerdata.py", line 203, in parse_config_file
    return bb.parse.handle(fn, data, include)
  File "/home/jancumps/iot2020/poky/bitbake/lib/bb/parse/__init__.py", line 113, in handle
    return h['handle'](fn, data, include)
  File "/home/jancumps/iot2020/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 124, in handle
    abs_fn = resolve_file(fn, data)
  File "/home/jancumps/iot2020/poky/bitbake/lib/bb/parse/__init__.py", line 136, in resolve_file
    raise IOError(errno.ENOENT, "file %s not found" % fn)
IOError: [Errno 2] file /home/build/poky/meta-intel/conf/layer.conf not found

ERROR: Unable to parse /home/build/poky/meta-intel/conf/layer.conf: [Errno 2] file /home/build/poky/meta-intel/conf/layer.conf not found
```

SaschaWeisenberger, 2017-04-09:
When you go to /home/jancumps/iot2020/iot2000build/conf/bblayers.conf, do you see paths beginning with "/home/build/..." or with "/home/jancumps/iot2020/poky/..."?

jancumps, 2017-04-09:
You have put me in the right direction. This is the content:

```
BBLAYERS ?= " \
  /home/jancumps/iot2020/poky/meta \
  /home/jancumps/iot2020/poky/meta-poky \
  /home/jancumps/iot2020/poky/meta-yocto-bsp \
  /home/build/poky/meta-intel \
  /home/build/poky/meta-iot2000/meta-iot2000-bsp \
```

I strictly only made the changes that were marked as green in the readme. I'll adapt it...

closed 2017-04-12 by jancumps