siemens/meta-iot2000 #15: BSP-Patch failed

Sneaker313, 2017-04-10:
I'm trying to build the example image for the iot2000. 
I used the morty-branch and followed the steps, described in the README file.
During the Build by bitbake, i get the following Error:

ERROR: linux-yocto-4.8.17+gitAUTOINC+bb6984f46b_9bcb4ea3fa-r0 do_patch: Function failed: do_patch (log file is located at /home/super/Dokumente/poky-morty-16.0.1/build/tmp/work/i586-nlp-32-intel-common-poky-linux/linux-yocto/4.8.17+gitAUTOINC+bb6984f46b_9bcb4ea3fa-r0/temp/log.do_patch.5552)
ERROR: Logfile of failure stored in: /home/super/Dokumente/poky-morty-16.0.1/build/tmp/work/i586-nlp-32-intel-common-poky-linux/linux-yocto/4.8.17+gitAUTOINC+bb6984f46b_9bcb4ea3fa-r0/temp/log.do_patch.5552

ERROR: Task (/home/super/Dokumente/poky-morty-16.0.1/meta/recipes-kernel/linux/linux-yocto_4.8.bb:do_patch) failed with exit code '1'

Any ideas how i can fix this?

Logfile:
DEBUG: Executing shell function do_patch
(1/18) 0001-stmmac-adding-support-for-platform-IOT2000.patch
(2/18) 0002-serial-uapi-Add-support-for-bus-termination.patch
(3/18) 0003-serial-8250_pci-Use-symbolic-constants-for-EXAR-s-MP.patch
(4/18) 0004-serial-8250_pci-Fix-EXAR-feature-control-register-co.patch
(5/18) 0005-serial-8250_pci-Add-support-for-IOT2000-platform.patch
[INFO]: check of .kernel-meta//patches//./0005-serial-8250_pci-Add-support-for-IOT2000-platform.patch with "git am" did not pass, trying reduced context.
(6/18) 0006-serial-8250_pci-Add-support-for-red-user-LED-on-IOT2.patch
[INFO]: check of .kernel-meta//patches//./0006-serial-8250_pci-Add-support-for-red-user-LED-on-IOT2.patch with "git am" did not pass, trying reduced context.
(7/18) 0007-efi-Move-efi_status_to_err-to-drivers-firmware-efi.patch
[INFO]: check of .kernel-meta//patches//./0007-efi-Move-efi_status_to_err-to-drivers-firmware-efi.patch with "git am" did not pass, trying reduced context.
[INFO]: Context reduced git-am of .kernel-meta//patches//./0007-efi-Move-efi_status_to_err-to-drivers-firmware-efi.patch with "git am" did not work, trying "apply".
error: patch failed: drivers/firmware/efi/vars.c:318
error: drivers/firmware/efi/vars.c: patch does not apply
Context reduced to (1/1) to apply fragment at 1001
[ERROR]: Application of .kernel-meta//patches//./0007-efi-Move-efi_status_to_err-to-drivers-firmware-efi.patch failed.
         Patch needs to be refreshed. Sample resolution script:
             .git/rebase-apply/resolve_rejects
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_patch (log file is located at /home/super/Dokumente/poky-morty-16.0.1/build/tmp/work/i586-nlp-32-intel-common-poky-linux/linux-yocto/4.8.17+gitAUTOINC+bb6984f46b_9bcb4ea3fa-r0/temp/log.do_patch.5552)

My Building Configurations:
Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "i586-poky-linux"
MACHINE           = "iot2000"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2.1"
TUNE_FEATURES     = "m32 i586-nlp"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "<unknown>:<unknown>"
meta-oe           = "HEAD:1efa5d623bc64659b57389e50be2568b1355d5f7"
meta-intel        = "morty:6add41510412ca196efb3e4f949d403a8b6f35d7"
meta-java         = "HEAD:a265b31ec7d022be254abdf959360a7624208585"
meta-intel-iot-middleware = "HEAD:fc8eabfa4fb54802d3f97123b9d2954450175e33"
meta-iot2000-bsp  
meta-iot2000-example = "<unknown>:<unknown>"

jan-kiszka, 2017-04-10:
Oh, the version in github still requires
```
PREFERRED_VERSION_linux-yocto = "4.4%"
```
in the local.conf. I'll push one later that makes this choice automatically.

Sneaker313, 2017-04-13:
Thank you, that solved my Problem :+1:

closed 2017-04-13 by Sneaker313