siemens/meta-iot2000 #16: failed to bitbake  iot2000-example-imag depends nodejs but not enabled

jancumps, 2017-04-10:
I'm trying to build the image iot2000-example-image,
by following [https://github.com/siemens/meta-iot2000/tree/master/meta-iot2000-example#sdk-target-architecture](https://github.com/siemens/meta-iot2000/tree/master/meta-iot2000-example#sdk-target-architecture) and where indicated also [https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-bsp/README.md](https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-bsp/README.md).

Once i'm bitbaking, I get this error immediately:
```
jancumps@jancumps-VirtualBox:~/iot2020/iot2000-build$ bitbake iot2000-example-image
ERROR: Layer 'iot2000-example' depends on layer 'nodejs', but this layer is not enabled in your configuration
```

WHere should I configure that?

jancumps, 2017-04-10:
I fixed this by adding the folowing line in build/conf/bblayers.conf, after meta-java:
`/home/build/poky/meta-nodejs \`

closed 2017-04-10 by jancumps

jan-kiszka, 2017-04-10:
Yes, ran into the same issue. I've queued a patch to adjust the README.