siemens/meta-iot2000 #19: failed to bitbake iot2000-example-image: compile error in task 407 MRAA

jancumps, 2017-04-12:
'm trying to build the image iot2000-example-image,
by following https://github.com/siemens/meta-iot2000/tree/master/meta-iot2000-example#sdk-target-architecture and where indicated also https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-bsp/README.md.

Once i'm bitbaking, I get this error when the MRAA libs get built:
`ERROR: Task 407 (/home/jancumps/iot2020/poky/meta-intel-iot-middleware/recipes-devtools/mraa/mraa_0.10.1.bb, do_compile) failed with exit code '1'`

`mraajsJAVASCRIPT_wrap.cxx:964:27: error: 'WeakCallbackData' in namespace 'v8' does not name a template type`

A number of errors occur with similar v8 namespace references. This is the first one:

 [ 62%] Built target mraa-i2c
| make -f examples/CMakeFiles/spi_max7219.dir/build.make examples/CMakeFiles/spi_max7219.dir/depend
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:964:27: error: 'WeakCallbackData' in namespace 'v8' does not name a template type
|    void (*dtor) (const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data);
|                            ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:964:43: error: expected ',' or '...' before '<' token
|    void (*dtor) (const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data);
|                                            ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1012:53: error: 'WeakCallbackData' in namespace 'v8' does not name a template type
|  SWIGRUNTIME void SWIGV8_Proxy_DefaultDtor(const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data) {
|                                                      ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1012:69: error: expected ',' or '...' before '<' token
|  SWIGRUNTIME void SWIGV8_Proxy_DefaultDtor(const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data) {
|                                                                      ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: In function 'void SWIGV8_Proxy_DefaultDtor(int)':
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1013:25: error: 'data' was not declared in this scope
|    SWIGV8_Proxy *proxy = data.GetParameter();
|                          ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: In function 'void SWIGV8_SetPrivateData(v8::Handle<v8::Object>, void*, swig_type_info*, int)':
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1084:78: error: no matching function for call to 'v8::Persistent<v8::Object>::SetWeak(SWIGV8_Proxy*&, void (*&)(int))'
|      cdata->handle.SetWeak(cdata, ((SWIGV8_ClientData*)info->clientdata)->dtor);
|                                                                               ^
| In file included from /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/node.h:42:0,
|                  from /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:169:
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:557:18: note: candidate: template<class P> void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = P; T = v8::Object]
|    V8_INLINE void SetWeak(P* parameter,
|                   ^
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:557:18: note:   template argument deduction/substitution failed:
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1084:78: note:   candidate expects 3 arguments, 2 provided
|      cdata->handle.SetWeak(cdata, ((SWIGV8_ClientData*)info->clientdata)->dtor);
|                                                                               ^
| In file included from /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/node.h:42:0,
|                  from /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:169:
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:568:18: note: candidate: void v8::PersistentBase<T>::SetWeak() [with T = v8::Object]
|    V8_INLINE void SetWeak();
|                   ^
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:568:18: note:   candidate expects 0 arguments, 2 provided
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1086:58: error: no matching function for call to 'v8::Persistent<v8::Object>::SetWeak(SWIGV8_Proxy*&, void (&)(int))'
|      cdata->handle.SetWeak(cdata, SWIGV8_Proxy_DefaultDtor);
|                                                           ^
| In file included from /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/node.h:42:0,
|                  from /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:169:
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:557:18: note: candidate: template<class P> void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = P; T = v8::Object]
|    V8_INLINE void SetWeak(P* parameter,
|                   ^
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:557:18: note:   template argument deduction/substitution failed:
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1086:58: note:   candidate expects 3 arguments, 2 provided
|      cdata->handle.SetWeak(cdata, SWIGV8_Proxy_DefaultDtor);
|                                                           ^
| In file included from /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/node.h:42:0,
|                  from /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:169:
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:568:18: note: candidate: void v8::PersistentBase<T>::SetWeak() [with T = v8::Object]
|    V8_INLINE void SetWeak();
|                   ^
| /home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h:568:18: note:   candidate expects 0 arguments, 2 provided
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: In function 'int SwigV8Packed_Check(v8::Handle<v8::Value>)':
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1240:40: error: 'class v8::Object' has no member named 'GetHiddenValue'
|    v8::Handle<v8::Value> flag = objRef->GetHiddenValue(SWIGV8_STRING_NEW("__swig__packed_data__"));
|                                         ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: At global scope:
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1290:58: error: 'WeakCallbackData' in namespace 'v8' does not name a template type
|  SWIGRUNTIME void _wrap_SwigV8PackedData_delete(const v8::WeakCallbackData<v8::Object, SwigV8PackedData> &data) {
|                                                           ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1290:74: error: expected ',' or '...' before '<' token
|  SWIGRUNTIME void _wrap_SwigV8PackedData_delete(const v8::WeakCallbackData<v8::Object, SwigV8PackedData> &data) {
[... cut ...]
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1320:8: error: 'class v8::Object' has no member named 'SetHiddenValue'
|    obj->SetHiddenValue(SWIGV8_STRING_NEW("__swig__packed_data__"), SWIGV8_BOOLEAN_NEW(true));
|         ^
| /home/jancumps/iot2020/iot2000-build/tmp/work/i586-nlp-32-poky-linux/mraa/1.5.1-r0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:1344:61: error: no matching function for call to 'v8::Persistent<v8::Object>::SetWeak(SwigV8PackedData*&, void (&)(int))'
|    cdata->handle.SetWeak(cdata, _wrap_SwigV8PackedData_delete);
|

jancumps, 2017-04-12:
Attached the error log for that job:

[log.do_compile.2323.txt](https://github.com/siemens/meta-iot2000/files/917456/log.do_compile.2323.txt)

jancumps, 2017-04-12:
and the v8 file is fetched as part of the build, located in:
/home/jancumps/iot2020/iot2000-build/tmp/sysroots/iot2000/usr/include/node/v8.h

jan-kiszka, 2017-04-13:
Could you share the startup report of bitbake about which layers in which revisions it is building? Just to check that there is no inconsistency.

jancumps, 2017-04-13:
Here's the startup report

```
WARNING: Host distribution "Ubuntu-16.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache...done.
Loaded 2215 entries from dependency cache.
Parsing recipes...done.
Parsing of 1646 .bb files complete (1645 cached, 1 parsed). 2215 targets, 106 skipped, 3 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 6.9.2 of nodejs not available (for item nodejs)
NOTE: versions of nodejs available: 0.10.48 0.12.18 0.8.28 4.2.4 4.4.3 4.8.0 6.10.0 7.7.3
NOTE: preferred version 6.9.2 of nodejs not available (for item nodejs-npm)
NOTE: versions of nodejs available: 0.10.48 0.12.18 0.8.28 4.2.4 4.4.3 4.8.0 6.10.0 7.7.3
NOTE: preferred version 6.9.2 of nodejs not available (for item nodejs)
NOTE: versions of nodejs available: 0.10.48 0.12.18 0.8.28 4.2.4 4.4.3 4.8.0 6.10.0 7.7.3
NOTE: preferred version 6.9.2 of nodejs-native not available (for item nodejs-native)
NOTE: versions of nodejs-native available: 0.10.48 0.12.18 0.8.28 4.2.4 4.4.3 4.8.0 6.10.0 7.7.3
NOTE: multiple providers are available for runtime java2-runtime (cacao, openjre-8, jamvm, openjdk-7-jre, openjdk-8)
NOTE: consider defining a PREFERRED_RPROVIDER entry to match java2-runtime
NOTE: preferred version 6.9.2 of nodejs not available (for item nodejs-dev)
NOTE: versions of nodejs available: 0.10.48 0.12.18 0.8.28 4.2.4 4.4.3 4.8.0 6.10.0 7.7.3
NOTE: multiple providers are available for virtual/java-native (cacao-native, jamvm-native)
NOTE: consider defining a PREFERRED_PROVIDER entry to match virtual/java-native
NOTE: multiple providers are available for virtual/java-initial-native (cacao-initial-native, jamvm-initial-native)
NOTE: consider defining a PREFERRED_PROVIDER entry to match virtual/java-initial-native

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "i586-poky-linux"
MACHINE           = "iot2000"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1"
TUNE_FEATURES     = "m32 i586-nlp"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "HEAD:6c1c01392d91f512e2949ad1d57a75a8077478ba"
meta-oe           = "HEAD:247b1267bbe95719cd4877d2d3cfbaf2a2f4865a"
meta-intel        = "HEAD:9a06fc5bce051c47816d139184753ecdcc29987c"
meta-java         = "HEAD:9edf7d5aa5bdcf9afc4365fd1d85feca72b4ef1a"
meta-nodejs       = "krogoth:78018dc7dc02b5039a165801d09c00564687a1b6"
meta-intel-iot-middleware = "HEAD:821cf14c8304669d9ce0c5b87b9be5a6eecff6e5"
meta-iot2000-bsp  
meta-iot2000-example = "master:f20ee85cbca87169b062cf28c135e032e87f8b9a"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Running task 2364 of 4504 (ID: 407, /home/jancumps/iot2020/poky/meta-intel-iot-middleware/recipes-devtools/mraa/mraa_0.10.1.bb, do_compile)

```

jan-kiszka, 2017-04-13:
Could you retry with the state that current master encodes? We've updated some layers. Just to avoid hunting an already resolve issue.

jancumps, 2017-04-13:
I'm sorry, I don't understand the question.
Do you want me to try without the ... git ... checkout #hexcode  commands, just the plain checkouts?

jan-kiszka, 2017-04-14:
We've updated the hashes and, thus, the layer revisions in the latest master version.

Just go to all the dependent layer checkout, run "git fetch origin", followed by "git checkout <new-hash>". Also update meta-iot2000 itself. Then rebuild.

jancumps, 2017-04-14:
I've done the sync of each layer and the build is ripping on...
No error at this moment. I'll update this issue with results.
For info, this is the version of the layers reported by bitbake after the git resync:
```
meta              
meta-poky         
meta-yocto-bsp    = "HEAD:6c1c01392d91f512e2949ad1d57a75a8077478ba"
meta-oe           = "HEAD:247b1267bbe95719cd4877d2d3cfbaf2a2f4865a"
meta-intel        = "HEAD:9a06fc5bce051c47816d139184753ecdcc29987c"
meta-java         = "HEAD:9edf7d5aa5bdcf9afc4365fd1d85feca72b4ef1a"
meta-nodejs       = "krogoth:78018dc7dc02b5039a165801d09c00564687a1b6"
meta-intel-iot-middleware = "HEAD:821cf14c8304669d9ce0c5b87b9be5a6eecff6e5"
meta-iot2000-bsp  
meta-iot2000-example = "master:a3d49e2c64f35e6806ce13036a79196609553e1a"
```

jancumps, 2017-04-14:
The build was successful. Thanks.
I'll now try to run it on the device...

closed 2017-04-14 by jancumps