siemens/meta-iot2000 #20: Analog Read doesn't work with meta-iot2000 default configuration

yetkinozturk, 2017-04-20:
gpio and pwm work properly but ADC chip driver is not initialized successfully. 
I have tested it with Intel's mraa examples on Analog read and with the method described on [http://www.malinov.com/Home/sergey-s-blog/intelgalileo-programminggpiofromlinux](http://www.malinov.com/Home/sergey-s-blog/intelgalileo-programminggpiofromlinux)

echo -n "37" > /sys/class/gpio/export
echo -n "out" > /sys/class/gpio/gpio37/direction
echo -n "0" > /sys/class/gpio/gpio37/value
but still,
there is no "/sys/bus/iio/devices/iio\:device0"

jan-kiszka, 2017-04-20:
Works for me 😄.
```
root@iot2000:~# ls /sys/bus/iio/devices/iio\:device0
buffer  in_voltage0_raw    in_voltage1_raw    in_voltage2_raw    in_voltage3_raw    in_voltage4_raw    in_voltage5_raw    in_voltage6_raw    in_voltage7_raw    name   scan_elements  trigger
dev     in_voltage0_scale  in_voltage1_scale  in_voltage2_scale  in_voltage3_scale  in_voltage4_scale  in_voltage5_scale  in_voltage6_scale  in_voltage7_scale  power  subsystem      uevent
```
Seriously: What is your setup? Which image version / git revision? Which device are you trying this on?

yetkinozturk, 2017-04-20:
:) oh, I have just built the image according to current README https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-bsp/README.md but i haven't tried it on my iot2020 do they have different ADC chips?

yetkinozturk, 2017-04-20:
just tried the same image on iot2020, it is OK. It just doesnt work on Intel Galileo Gen2 sorry to bother, closing the issue.

closed 2017-04-20 by yetkinozturk

jan-kiszka, 2017-04-20:
Ah! Yeah, they have a nice firmware bug on that board. It's fixed with a ACPI table patch that comes with yocto. Found that by chance once some time back.

yetkinozturk, 2017-04-20:
That explains everything. Thanks!