siemens/meta-iot2000 #24: Suggestion: NTP Client as IoT2020 has no battery backup.

DanPlastina, 2017-05-07:
It would be fantastic if an NTP client was added to facilitate setting the clock on power up.

nospam2000, 2017-05-08:
Adding ntpd will be one of my first projects for a self-baked image.

jan-kiszka, 2017-05-08:
Busybox has a built-in service for this, we just lack its activation. But we are taking patches 😉.

nospam2000, 2017-05-08:
BusyBox's ntp client is small, but also has limitations. A short look at its [REAME](http://doolittle.icarus.com/ntpclient/README) says for example that is doesn't support multiple servers.
ntpd is also not only an ntp client but also a ntp server.

Nevertheless BusyBox will be appropriate for many users.

jan-kiszka, 2017-05-08:
Hmm, the code says the busybox client supports multiple servers. It also has an optional server mode as well, but I have no own experience to compare it to a full-blown one.

nospam2000, 2017-05-20:
I have now ntpd running using the standard recipe from OpenEmbedded and an adapted config file. Do you want a patch or merge request or a description how to get it into the image?

jan-kiszka, 2017-05-21:
Pull request, please. Regarding the style of the commit, follow other commits in this repo. The log should describe why we want this change and carry a signed-off according to https://developercertificate.org.

I would suggest to preconfigure things generically for ntpclient operation, using e.g. the parameters of http://www.pool.ntp.org/en/use.html.

Did you also try the busybox applet? In any case, please document the reason for choosing the full-featured one.

nospam2000, 2017-05-31:
sorry, I won't submit a pull request, because I quit using yocto and switch to debian.

jan-kiszka, 2017-05-31:
...at your own risk.
You will have to rebuild an undefined number of packages then (I think I cited a war story elsewhere). An own kernel will also be required unless you can even live without network (will improve with 4.12 and then significantly with 4.13).
If you want to do this properly, this can be an interesting tool to describe the process: https://github.com/ilbers/isar. We didn't follow this path for the IOT2000 due to the significant effort + the fact that users will still not be able to pull from arbitrary Debian repos.

jan-kiszka, 2017-06-01:
Feature merged - thanks for the suggestion and inputs.

closed 2017-06-01 by jan-kiszka