siemens/meta-iot2000 #25: Build environment

uwburn, 2017-05-09:
Hello, i'm trying to build the latest example-image from the repo master (i already did it successfully in the past), and i'm having some issue with task do_image_ext4 (getting error "mkfs.ext4: No data available while populating file system"). I have already google for that error but have found no answer about it, before spending several days after it i would try to start with a clean build environment (i played quite bit with it). 

What Linux distro and version are you using to build the image? Poky is complaining with a warning that Ubuntu 16.04 is not a validated version, as long as everithing works i just ignore it, but from past experience i know that it might cause problems hard to track down.

Thanks in advance.

jan-kiszka, 2017-05-09:
Standard questions:
- does deleting <build-folder>/tmp help?
- does cleaning the complete build folder help (long rebuild)?
- what is your configuration as reported by bitbake during recipe parsing?

FWIW, I'm building on SUSE Leap 42.2, and I receive a complaint as well. But it works.

uwburn, 2017-05-09:
Thanks for quick response.
Haven't tried yet to clean the build (it's a virtual machine on a laptop, so it takes quite a long time to do all the build), but i will.

Anyway, a more or less a month ago i had to bake a custom image to update node to v6, and with Ubuntu 16.10 i had a lot of issues in completing the java layer tasks... in the end, the only way out was to downgrade to 16.04.

uwburn, 2017-05-09:
Deleting /tmp did the trick, the image was built successfully.
Thanks for the tip!

jan-kiszka, 2017-05-09:
You are welcome!

closed 2017-05-09 by jan-kiszka