siemens/meta-iot2000 #35: Add support for defining time zone via iot2000setup

jan-kiszka, 2017-06-05:
Possibly needed steps:
- list available zones from /usr/share/zoneinfo/
- write selected zone to /etc/timezone
- ln -s /usr/share/zoneinfo/<zone> /etc/localtime

nospam2000, 2017-06-09:
you need the recipe "[tzdata](https://layers.openembedded.org/layerindex/recipe/39612/)", then you can do
`ln -s -f /usr/share/zoneinfo/Europe/Paris /etc/localtime`

There are only a few timezones contained in /usr/share/zoneinfo, e.g. no "Berlin"
```
# ls -lR /usr/share/zoneinfo/Europe/
-rw-r--r-- 1 root root 3687 Apr 25 18:52 London
-rw-r--r-- 1 root root 1544 Apr 25 18:52 Moscow
-rw-r--r-- 1 root root 2971 Jun  9 18:02 Paris
```

tzselect from recipe "[tzcode](https://layers.openembedded.org/layerindex/recipe/5313/)" doesn't work:
```
# tzselect
/usr/bin/tzselect: line 180: /usr/share/zoneinfo/zone1970.tab: No such file or directory
/usr/bin/tzselect: time zone files are not set up correctly
```

jan-kiszka, 2017-06-09:
We have tzdata already, see [here](https://github.com/siemens/meta-iot2000/commit/8041b7107be06dfde8f2fcba4581eda2773c13cf#diff-8079e8d037739991c2b21cec4df3e00fR17). This seems like sparse Yocto furnishing...

Didn't look into tzcode yet. Maybe fixable.

jan-kiszka, 2017-06-16:
Missing additional tzdata packages on our side plus an upstream oe-core bug. Both addressed now, tzselect works. However, the proper solution remains setting those two files.

commit 3e23ebc3 (2019-07-12): example: iot2000setup: Implement timezone dialog

Closes #35.
closed 2019-07-12 by BaochengSu