siemens/meta-iot2000 #39: Integrate in-field image update support

jan-kiszka, 2017-06-05:
The bits and pieces for a software update mechanism for IOT2000 devices is work in progress. It would be useful to integrate this feature in form of a reference implementation into meta-iot2000 once the missing pieces are done and published.

jan-kiszka, 2017-11-06:
Building blocks:
- https://github.com/siemens/meta-efibootguard
- https://github.com/sbabic/meta-swupdate

nuno407, 2018-08-23:
Hi @jan-kiszka ,
Any update regarding this issue?
Thank you, Nuno

nuno407, 2018-08-24:
Hi @jan-kiszka,
Thank you for the fast response. I think this is a major feature.
How can I help in order to accomplish it?

jan-kiszka, 2018-08-24:
@nuno407 Thanks a lot. However, the immediate to-dos at this point are hard to break down, but we would come back to you for testing: I'm optimistic that we can pick this up again in couple of weeks.

jan-kiszka, 2019-05-29:
Merged into master now.

closed 2019-05-29 by jan-kiszka