siemens/meta-iot2000 #42: Building the image on Fedora 25 fails due to missing acpica-native

Flonix, 2017-06-07:
Hi!

I'm trying to build the image on a Fedora 25 system.
The build fails with the following error message:

```
$ bitbake core-image-minimal
WARNING: Host distribution "Fedora-25" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |###################################################################################################| Time: 0:00:00
Loaded 1343 entries from dependency cache.
Parsing recipes: 100% |#################################################################################################| Time: 0:00:00
Parsing of 888 .bb files complete (886 cached, 2 parsed). 1344 targets, 56 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'acpica-native' (but /home/flo/workspace/poky/meta-iot2000/meta-iot2000-bsp/recipes-bsp/acpi-upgrades/acpi-upgrades.bb DEPENDS on or otherwise requires it). Close matches:
  cpio-native
  byacc-native
  asciidoc-native
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'core-image-minimal-initramfs', 'acpi-upgrades', 'acpica-native']
```

acpica-native is required by `meta-iot2000/meta-iot2000-bsp/recipes-bsp/acpi-upgrades/acpi-upgrades.bb` 

Who should provide acpica-native?
Is there something missing on my build system?

SaschaWeisenberger, 2017-06-07:
You are right, this recipe is missing. It is in meta-oe, which we refer to in the example image. Fix is on the way...

jan-kiszka, 2017-06-08:
Yeah, by bad. It's currently a simple documentation deficit.

I already have a configuration artifact for scripted setup available in our Intranet. Drop me a note for a link. That stuff will be merged here once we are through with the OSS release process, and a somewhat stable version of that new tool is publicly available.

closed 2017-06-08 by jan-kiszka