siemens/meta-iot2000 #47: boot error: kernel panic rootfs.

malconxx, 2017-06-14:
I built the image and recorded it on the memory card.
During the boot process this error occurs:

VFS: Unable to mount root fs on unknown-block(0,0)
User configuration error - no valid root filesystem found
Kernel panic - not syncing: Invalid configuration from end user prevents continuing
CPU: 0 PID: 1 Comm: swapper Not tainted 4.8.17-yocto-standard #1
Hardware name: Intel Corp. QUARK/SIMATIC IOT2000, BIOS V24.01.01 01/01/2014
 f5105e00 f5105e00 f5105ec4 c13789f9 f5105edc c10f1e7b f68e3000 f5105ef4
 f68e3000 fffffffa f5105f20 c1938f1c c17a9fb8 00008000 c17a9893 f5fc6378
 6e6b6e75 2d6e776f 636f6c62 2c30286b c1002930 00006180 00000000 f5105f38
Call Trace:
 [<c13789f9>] dump_stack+0x16/0x1d
 [<c10f1e7b>] panic+0x86/0x18b
 [<c1938f1c>] mount_block_root+0x132/0x1da
 [<c1002930>] ? x86_pmu_commit_txn.part.5+0x50/0x80
 [<c193916b>] mount_root+0xef/0xf6
 [<c1939bc0>] ? initrd_load+0x282/0x28e
 [<c19392b7>] prepare_namespace+0x145/0x176
 [<c1938c61>] kernel_init_freeable+0x1b8/0x1ca
 [<c16a4ef0>] kernel_init+0x10/0x100
 [<c106bb51>] ? schedule_tail+0x11/0x50
 [<c16a91ee>] ret_from_kernel_thread+0xe/0x30
 [<c16a4ee0>] ? rest_init+0x80/0x80
Kernel Offset: disabled
---[ end Kernel panic - not syncing: Invalid configuration from end user prevents continuing
mmcblk0: mmc0:0007 SL16G 14.5 GiB

jan-kiszka, 2017-06-14:
Which revision of meta-iot2000 did you build? None of ours should select the Yocto 4.8 kernel that was booted on your system. The patches we still carry will not apply to it, but those alone do not explain the failing boot.

Also double-check again that you checked out the documented revisions of the layers we depends on, specifically meta-intel.

malconxx, 2017-06-19:
1) Machine used to generate the image:
- Fedora 25: Linux Kernel 4.11.5-200.fc25.x86_64 x86_64 GNU/Linux

2) About the poky version and the other features I am using the same patches described in the README. **Follow the poky log**:

[mabs@euclides iot2000-build]$ bitbake core-image-minimal
WARNING: Host distribution "Fedora-25" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |###################################################################| Time: 0:00:00
Loaded 2094 entries from dependency cache.
Parsing recipes: 100% |#################################################| Time: 0:00:03
Parsing of 1523 .bb files complete (1519 cached, 4 parsed). 2100 targets, 107 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "i586-poky-linux"
MACHINE           = "iot2000"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2.1"
TUNE_FEATURES     = "m32 i586-nlp"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "HEAD:6a1f33cc40bfac33cf030fe41e1a8efd1e5fad6f"
meta-oe           = "HEAD:fe5c83312de11e80b85680ef237f8acb04b4b26e"
meta-intel        = "HEAD:6add41510412ca196efb3e4f949d403a8b6f35d7"
meta-iot2000-bsp  = "HEAD:f20ee85cbca87169b062cf28c135e032e87f8b9a"
meta-golang       = "master:792695ff134bb81194cb11ae7b46c91b818485c4"

Initialising tasks: 100% |####################################################################| Time: 0:00:04
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 2335 tasks of which 2335 didn't need to be rerun and all succeeded.

Summary: There was 1 WARNING message shown.
[mabs@euclides iot2000-build]$

jan-kiszka, 2017-06-19:
Revision f20ee8 is the official release 2.1.3, but that required Yocto 2.1. If you want to build against 2.2, update to current master. If you want to reproduce that old image, follow the README describing the dependencies exactly.

Also note the new way of building the image without having all those manual checkout steps, see https://support.industry.siemens.com/tf/ww/en/posts/building-fresh-iot2000-sd-card-image-via-a-single-command/168681/?page=0&pageSize=10

malconxx, 2017-06-19:
I found the error. I modified the /efi/loader/entries/boot.conf and added at the end of the line the parameter "rootwait".

When I generated the image the file comes by default with this setting:
Options LABEL = Boot root = / dev / mmcblk0p2 console = ttyS1,115200n8 reboot = efi, warm rw LABEL = boot debugshell = 5

When the system was booted, the kernel panic error occurred because it could not mount the root file system.

jan-kiszka, 2017-06-19:
Yes, you had to add this parameter manually because of using the old version. It's fixed now in master, see 4b3e3f805d.

closed 2017-06-19 by jan-kiszka