siemens/meta-iot2000 #51: Example image build fails with nodejs

santiago-vanoli, 2017-06-23:
Hi, I'm trying to build the example image but keep getting this error:

ERROR: nodejs-native-6.10.2-r1.8 do_configure: Function failed: do_configure (log file is located at /home/santiago/iot2000-build/tmp/work/x86_64-linux/nodejs-native/6.10.2-r1.8/temp/log.do_configure.24361)
ERROR: Logfile of failure stored in: /home/santiago/iot2000-build/tmp/work/x86_64-linux/nodejs-native/6.10.2-r1.8/temp/log.do_configure.24361
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| /home/santiago/iot2000-build/tmp/work/x86_64-linux/nodejs-native/6.10.2-r1.8/temp/run.do_configure.24361: 108: /home/santiago/iot2000-build/tmp/work/x86_64-linux/nodejs-native/6.10.2-r1.8/temp/run.do_configure.24361: ./configure: Permission denied
| WARNING: exit code 126 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/santiago/iot2000-build/tmp/work/x86_64-linux/nodejs-native/6.10.2-r1.8/temp/log.do_configure.24361)
ERROR: Task (virtual:native:/home/santiago/poky/meta-nodejs/recipes-devtools/nodejs/nodejs_6.10.2.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 803 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/santiago/poky/meta-nodejs/recipes-devtools/nodejs/nodejs_6.10.2.bb:do_configure
Summary: There were 3 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Any comment on where the error could be would be appreciated.

jan-kiszka, 2017-06-23:
Could you describe how (command steps) and what (early part of log shows layers and their revisions) you built?

santiago-vanoli, 2017-06-23:
The commands (in order) were:
$ git clone git://git.yoctoproject.org/poky.git poky -b morty
$ git clone git://git.yoctoproject.org/meta-intel poky/meta-intel -b morty
$ git clone http://git.openembedded.org/meta-openembedded poky/meta-openembedded
$ git -C poky checkout 6a1f33cc40bf
$ git -C poky/meta-intel checkout 6add41510412
$ git -C poky/meta-openembedded checkout fe5c83312de1
$ git clone git://github.com/siemens/meta-iot2000 poky/meta-iot2000
$ git clone git://git.openembedded.org/meta-openembedded poky/meta-oe -b morty
$ git clone git://git.yoctoproject.org/meta-intel-iot-middleware poky/meta-intel-iot-middleware -b master
$ git clone git://github.com/imyller/meta-nodejs.git poky/meta-nodejs -b morty
$ git -C poky/meta-oe checkout fe5c83312de1
$ git -C poky/meta-intel-iot-middleware checkout fc8eabfa4fb5
$ git -C poky/meta-nodejs checkout 57e534dd8a53
$ source poky/oe-init-build-env iot2000-build
Then, modified iot2000-build/conf/bblayers.conf and iot2000-build/conf/local.conf as described in https://github.com/siemens/meta-iot2000/tree/master/meta-iot2000-example

Early par of log is:

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "i586-poky-linux"
MACHINE           = "iot2000"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2.1"
TUNE_FEATURES     = "m32 i586-nlp"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "HEAD:6a1f33cc40bfac33cf030fe41e1a8efd1e5fad6f"
meta-oe           = "HEAD:fe5c83312de11e80b85680ef237f8acb04b4b26e"
meta-intel        = "HEAD:6add41510412ca196efb3e4f949d403a8b6f35d7"
meta-intel-iot-middleware = "HEAD:fc8eabfa4fb54802d3f97123b9d2954450175e33"
meta-nodejs       = "HEAD:57e534dd8a536a1ce8310d5689258c201ef84652"
meta-iot2000-bsp  
meta-iot2000-example = "master:6a074112a12d4e2c9ca6c885bc85b70436b572fc"

jan-kiszka, 2017-06-23:
Yeah, that's different from what we are testing here ATM - README and real setups diverged:

```
2017-06-23 10:41:05 - INFO     - Build Configuration:
2017-06-23 10:41:05 - INFO     - BB_VERSION        = "1.32.0"
2017-06-23 10:41:05 - INFO     - BUILD_SYS         = "x86_64-linux"
2017-06-23 10:41:05 - INFO     - NATIVELSBSTRING   = "Debian-8.8"
2017-06-23 10:41:05 - INFO     - TARGET_SYS        = "i586-poky-linux"
2017-06-23 10:41:05 - INFO     - MACHINE           = "iot2000"
2017-06-23 10:41:05 - INFO     - DISTRO            = "poky"
2017-06-23 10:41:05 - INFO     - DISTRO_VERSION    = "2.2.2"
2017-06-23 10:41:05 - INFO     - TUNE_FEATURES     = "m32 i586-nlp"
2017-06-23 10:41:05 - INFO     - TARGET_FPU        = ""
2017-06-23 10:41:05 - INFO     - meta-intel        = "HEAD:6add41510412ca196efb3e4f949d403a8b6f35d7"
2017-06-23 10:41:05 - INFO     - meta
2017-06-23 10:41:05 - INFO     - meta-yocto-bsp
2017-06-23 10:41:05 - INFO     - meta-poky         = "HEAD:21aa71dbaaea0aaf84329c0181dfac91f6830573"
2017-06-23 10:41:05 - INFO     - meta-iot2000-bsp
2017-06-23 10:41:05 - INFO     - meta-iot2000-example = "HEAD:ae0ff0336c170d15dfce8007a57042067e0d07a4"
2017-06-23 10:41:05 - INFO     - meta-nodejs       = "master:eec531e97a17bfd406f3bf76dee4057dcf5286a4"
2017-06-23 10:41:05 - INFO     - meta-intel-iot-middleware = "master:fc8eabfa4fb54802d3f97123b9d2954450175e33"
2017-06-23 10:41:05 - INFO     - meta-oe           = "HEAD:fe5c83312de11e80b85680ef237f8acb04b4b26e"
```

There is this new tool kas (https://github.com/siemens/kas) and a configuration file for the example image in the jan/kas branch. You could try that path directly, or at least pick up the revisions from it? We will soon switch the doc to that method as well.

malconxx, 2017-06-23:
using this script ! work fine

#!/bin/bash

# poky
git clone git://git.yoctoproject.org/poky.git poky -b morty
# meta-iot2000
git clone git://github.com/siemens/meta-iot2000 poky/meta-iot2000 -b master
git clone git://git.openembedded.org/meta-openembedded poky/meta-oe -b morty
git clone git://git.yoctoproject.org/meta-intel-iot-middleware poky/meta-intel-iot-middleware -b master
git clone git://github.com/imyller/meta-nodejs.git poky/meta-nodejs -b morty

git -C poky/meta-oe checkout fe5c83312de1
git -C poky/meta-intel-iot-middleware checkout fc8eabfa4fb5
git -C poky/meta-nodejs checkout 57e534dd8a53

# meta-intel
git clone git://git.yoctoproject.org/meta-intel poky/meta-intel -b morty

jan-kiszka, 2017-06-23:
@malconxx, that looks like it is referencing the same revisions @santiago-vanoli was trying already. Maybe there is an issue, depending on the host setup, with that set. Therefore my suggestion to use the kas build environment or at least the partly newer revisions from the meta-iot2000-example/kas-iot2000-example-image.yml.

santiago-vanoli, 2017-07-13:
Tried to build in another host, didn't have a problem. It was probably a problem with my host machine. 
Thanks for the assist!

closed 2017-07-13 by santiago-vanoli