siemens/meta-iot2000 #52: bitbake fails on Ubuntu 17.04

jcgruenhage, 2017-07-20:
I just tried to build the meta-iot2000-bsp image on Ubuntu 17.04, exactly like specified in the README, and it fails with this error message:

[image]

I also wanted to add that all of the tested distros listed on the README page are out of date, except OpenSuse 42.2, which will go out of date in less than a week. I'll test it with Debian 9 now, hopefully that will work.

jan-kiszka, 2017-07-20:
The REAME in master is likely out of date and will be replaced soon with what is in jan/staging. Specifically the Docker way can be interesting if you face compatibility/dependency issues with your Distro (and will replace that list of tested distros).

jcgruenhage, 2017-07-20:
Oh, I hadn't even noticed the other branches.. I'll check out the docker build, thank you :)

jcgruenhage, 2017-07-20:
So, the docker build has worked mostly fine, but there is one thing that didn't quite work. Why is the user set to `id -u` in the container? Docker commands are usually run as root [(comment by RedHat why this should stay that way)](https://www.projectatomic.io/blog/2015/08/why-we-dont-let-non-root-users-run-docker-in-centos-fedora-or-rhel/), which causes the image to not run, because on startup it then complains, that it can't create a the root user because that already exists.

jan-kiszka, 2017-07-20:
Other distros are less strict, so things were modeled around them so far. But we should fix the image invocation as root in that container, ie. in [kas](https://github.com/siemens/kas). Thanks for the note!

jan-kiszka, 2017-07-21:
Patch for kas was sent upstream. So the Docker issue you reported will be addressed when we use the next version of kas (I'm waiting for it to merge a properly adjusted README).

closed 2017-07-21 by jan-kiszka