siemens/meta-iot2000 #55: iot2000setup does not remove node-red / mosquitto from autostart

toffe82, 2017-08-02:
If node-red was selected in autostart, deselecting it does not remove it from autostart, it will still start at the next reboot.
it is missing a parameter in the function to do that: 
- you have to add the option -f to update-rc.d  (line 425), do not forget to let a space between f and the double quote :
elif (status == "off"):
subprocess.call("update-rc.d -f " + fileName + " remove", shell=True, stdout=open(os.devnull, 'wb'))
os.remove("/etc/init.d/" + fileName)
The option is there for the removal of SSH at startup , but not there for node-red and mosquitto which share the same lines of code.

jan-kiszka, 2017-08-02:
Relates to #36: with proper scripts for them and generalization of the path taken for ssh already, we'll fix that implicitly.

jan-kiszka, 2019-05-28:
addressed in master now.

closed 2019-05-28 by jan-kiszka