siemens/meta-iot2000 #59: Please use a locale setting which supports utf-8.

BaochengSu, 2017-10-13:
When building bsp using "kas build meta-iot2000-bsp/kas.yml", below error showed up:

> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-iot2000-bsp$ /usr/bin/git rev-parse --show-toplevel
> 2017-10-13 23:54:38 - INFO     - Using /home/iot/iot2000-github/meta-iot2000 as root for repository meta-iot2000
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-intel$ /usr/bin/git cat-file -t 6add41510412ca196efb3e4f949d403a8b6f35d7
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/poky$ /usr/bin/git cat-file -t 7d5822bf4cb2089687c3c9a908cf4a4ef4e9a53a
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-openembedded$ /usr/bin/git cat-file -t fe5c83312de11e80b85680ef237f8acb04b4b26e
> 2017-10-13 23:54:38 - INFO     - Repository meta-intel already contains 6add41510412ca196efb3e4f949d403a8b6f35d7 as commit
> 2017-10-13 23:54:38 - INFO     - Repository poky already contains 7d5822bf4cb2089687c3c9a908cf4a4ef4e9a53a as commit
> 2017-10-13 23:54:38 - INFO     - Repository meta-openembedded already contains fe5c83312de11e80b85680ef237f8acb04b4b26e as commit
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-iot2000-bsp$ /usr/bin/git rev-parse --show-toplevel
> 2017-10-13 23:54:38 - INFO     - Using /home/iot/iot2000-github/meta-iot2000 as root for repository meta-iot2000
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-intel$ /usr/bin/git diff --shortstat
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-intel$ /usr/bin/git rev-parse --verify HEAD
> 2017-10-13 23:54:38 - INFO     - 6add41510412ca196efb3e4f949d403a8b6f35d7
> 2017-10-13 23:54:38 - INFO     - Repo meta-intel has already checkout out correct refspec. nothing to do
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/poky$ /usr/bin/git diff --shortstat
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/poky$ /usr/bin/git rev-parse --verify HEAD
> 2017-10-13 23:54:38 - INFO     - 7d5822bf4cb2089687c3c9a908cf4a4ef4e9a53a
> 2017-10-13 23:54:38 - INFO     - Repo poky has already checkout out correct refspec. nothing to do
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-openembedded$ /usr/bin/git diff --shortstat
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-openembedded$ /usr/bin/git rev-parse --verify HEAD
> 2017-10-13 23:54:38 - INFO     - fe5c83312de11e80b85680ef237f8acb04b4b26e
> 2017-10-13 23:54:38 - INFO     - Repo meta-openembedded has already checkout out correct refspec. nothing to do
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-iot2000-bsp$ /usr/bin/git rev-parse --show-toplevel
> 2017-10-13 23:54:38 - INFO     - Using /home/iot/iot2000-github/meta-iot2000 as root for repository meta-iot2000
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/poky$ /tmp/tmp78md2pru /home/iot/iot2000-github/meta-iot2000/build
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/meta-iot2000-bsp$ /usr/bin/git rev-parse --show-toplevel
> 2017-10-13 23:54:38 - INFO     - Using /home/iot/iot2000-github/meta-iot2000 as root for repository meta-iot2000
> 2017-10-13 23:54:38 - INFO     - /home/iot/iot2000-github/meta-iot2000/build$ /home/iot/iot2000-github/meta-iot2000/poky/bitbake/bin/bitbake -k core-image-minimal -c build
> 2017-10-13 23:54:39 - ERROR    - Please use a locale setting which supports utf-8.
> 2017-10-13 23:54:39 - ERROR    - Python can't change the filesystem locale after loading so we need a utf-8 when python starts or things won't work.
> 2017-10-13 23:54:39 - ERROR    - Command "/home/iot/iot2000-github/meta-iot2000/build$ /home/iot/iot2000-github/meta-iot2000/poky/bitbake/bin/bitbake -k core-image-minimal -c build" failed
> --- Error summary ---
> Please use a locale setting which supports utf-8.
> Python can't change the filesystem locale after loading so we need a utf-8 when python starts or things won't work.

The output of command "printenv":

> iot@XXXXXXXXX:~/iot2000-github/meta-iot2000$ printenv
> LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
> HTTP_PROXY=http://192.168.100.100:1080/
> LANG=en_HK.UTF-8
> GDM_LANG=en_HK.utf8
> DISPLAY=:0
> GTK_OVERLAY_SCROLLING=0
> COLORTERM=truecolor
> XDG_VTNR=7
> SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
> XDG_SESSION_ID=14
> XDG_GREETER_DATA_DIR=/var/lib/lightdm/data/iot
> USER=iot
> DESKTOP_SESSION=lightdm-xsession
> QT_QPA_PLATFORMTHEME=gtk2
> PWD=/home/iot/iot2000-github/meta-iot2000
> HOME=/home/iot
> SSH_AGENT_PID=3957
> QT_ACCESSIBILITY=1
> XDG_SESSION_TYPE=x11
> XDG_DATA_DIRS=/usr/share/gnome:/usr/local/share/:/usr/share/
> http_proxy=http://192.168.100.100:1080/
> XDG_SESSION_DESKTOP=lightdm-xsession
> no_proxy=localhost,127.0.0.0/8,::1
> GTK_MODULES=gail:atk-bridge
> NO_PROXY=localhost,127.0.0.0/8,::1
> TERM=xterm-256color
> SHELL=/bin/bash
> VTE_VERSION=4601
> XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
> XDG_CURRENT_DESKTOP=X-Cinnamon
> GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
> QT_LINUX_ACCESSIBILITY_ALWAYS_ON=1
> SHLVL=1
> XDG_SEAT=seat0
> LANGUAGE=en_HK:en
> WINDOWID=37748742
> GDMSESSION=lightdm-xsession
> GNOME_DESKTOP_SESSION_ID=this-is-deprecated
> LOGNAME=iot
> DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
> XDG_RUNTIME_DIR=/run/user/1000
> XAUTHORITY=/home/iot/.Xauthority
> XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session5
> PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
> SESSION_MANAGER=local/IotBuildMachine:@/tmp/.ICE-unix/3929,unix/IotBuildMachine:/tmp/.ICE-unix/3929
> _=/usr/bin/printenv
> OLDPWD=/home/iot/iot2000-github/meta-iot2000/meta-iot2000-bsp
>

jan-kiszka, 2017-10-13:
What's your build environment (distro, version etc)? Does Yocto claim to support it? This seems like a tuning issue of the build environment, and even stock Yocto builds should be affected. The kas docker build should work, though.

BaochengSu, 2017-10-14:
I am using a new installed Debian 9.2 amd64 with:

> LANG=en_HK.UTF-8
> LC_ALL=
> LANGUAGE=en_HK:en

However, it seems Python3 doesn't recognized this kind of combination as UTF-8 encoding, the error message was happened here:
```
if sys.getfilesystemencoding() != "utf-8":
    sys.exit("Please use a locale setting which supports utf-8.\nPython can't change the filesystem locale after loading so we need a utf-8 when python starts or things won't work.")
```
from _meta-iot2000/poky/bitbake/bin/bitbake_ source file.

Anyway, after I used "_dpkg-reconfigure locales_" tool to regenerated the "en_US.UTF-8" locale and set it as the default locale, the problem is gone.

jan-kiszka, 2017-11-06:
Should we do anything further regarding this or close the issue?

BaochengSu, 2017-11-06:
I will close it.

closed 2017-11-06 by BaochengSu