siemens/meta-iot2000 #62: meta-intel-iot-middleware This layer is no longer maintained

danielycom, 2017-11-13:
meta-iot2000/meta-iot2000-example/kas.yml
meta-intel-iot-middleware
> This layer is no longer maintained - the useful recipes that used to be here have now been moved out to other layers (meta-oe, meta-python, meta-networking etc.)

So I dont need it anymore?

jan-kiszka, 2017-12-11:
Things moved on to meta-oe from meta-openembedded. We will switch when updating to Yocto 2.4, see https://github.com/siemens/meta-iot2000/tree/jan/rocko.

arfoll, 2017-12-11:
Sorry I missed this - yes pull it from the oe layers, as long as you don't
need java it works well.

meta-iot2000/tree/jan/rocko.

jan-kiszka, 2017-12-11:
Yeah, we dropped Java long ago - bloated the image and was breaking the build too often.

jan-kiszka, 2018-06-22:
Resolved in master by going to rocko.

closed 2018-06-22 by jan-kiszka