siemens/meta-iot2000 #66: Update to rocko (Yocto 2.4)

jan-kiszka, 2017-12-11:
Patches are there (https://github.com/siemens/meta-iot2000/tree/jan/rocko) but need further testing and likely have to wait for an earlier release of the morty baseline in current master.

The iot2000setup tool should be cleaned up along this, specifically addressing #36.

debevv, 2018-01-10:
Hi, I have a question regarding the update: I noticed that acpi-upgrades.cpio is generated but it's not included in the final image. Is there a problem in my build? Or it is not needed anymore?

jan-kiszka, 2018-01-11:
It is included here, just stored now at a different location (/efi/acpi-upgrades-iot2000.cpio). Did you see that the SSDT is no longer overloaded during boot (the kernel reports this, see dmesg)?

BTW, the acpi-upgrades will disappear with a BIOS update that is under preparation.

debevv, 2018-01-11:
I think the upgrade to rocko broke something in my application layer, I don't even get the EFI executable of GRUB now (we use it instead of systemd-boot). I'll answer your question when I manage to get a working build

jan-kiszka, 2018-01-11:
I removed grub dependencies more systematically, see https://github.com/siemens/meta-iot2000/commit/8b6a2251b5429c5865fe9dd2ac94004c5cc01c32, possibly a reason. Your layer will have to re-add all needed dependencies for that.

BTW, if you are adding grub in order to do firmware updates: We are also working on publishing a better (specifically safer) alternative to that, using EFI Boot Guard as bootloader. Stay tuned.

debevv, 2018-01-11:
Yes! Adding WKS_FILE_DEPENDS_BOOTLOADERS_x86 in my layer.conf fixed the problem with GRUB. 
Still no acpi-upgrades generated, even after a full rebuild. There are some warning and errors about ACPI in dmesg, but the system is working properly.

About EFI Boot Guard, is there an ETA on that? And what about swupdate integration?

jan-kiszka, 2018-01-11:
@andi8086 and @stormc are on this and can possibly provide some rough estimate.

If you share your build log with me, I could compare to a working one here.

stormc, 2018-01-12:
I just submitted the SWUpdate integration of EFI Boot Guard to the [swupdate mailinglist](https://groups.google.com/d/msg/swupdate/n7LaW4cBdFY/U2J-kJtNAQAJ), so it's on its way now...

stormc, 2018-01-15:
Just to keep you posted, EFI Boot Guard integration in SWUpdate has landed in [master](https://github.com/sbabic/swupdate/commit/07bf41de89d667f08722cfab50a306fdfedeb253).

debevv, 2018-01-15:
Here are my build logs (they are split because I ran out of RAM during the build).
[20180115091140.log](https://github.com/siemens/meta-iot2000/files/1631302/20180115091140.log)
[20180115101826.log](https://github.com/siemens/meta-iot2000/files/1631303/20180115101826.log)

EFI Boot Guard looks very interesting now, I'll see if we are still on time to replace GRUB with it in our system.

andi8086, 2018-01-15:
> @andi8086 and @stormc are on this and can possibly provide some rough estimate.
> 
> If you share your build log with me, I could compare to a working one here.
Sorry for delayed answer. Since my focus was on efibootguard, I did not
save my build logs before patching out stuff I was not interested in
that time. So it is easier that I have a look at your rocko branch now
and base my patches on yours to avoid duplicate work. Mine was almost
entirely for efibootguard anyhow.

Andreas

floion, 2018-03-16:
Hi,
are there plans for a rocko branch?
I see there's a 
> jan/rocko

branch, but is that a stable/final branch?

Thank you

jan-kiszka, 2018-03-17:
jan/rocko still requires some more testing to become master. it's scheduled after a release of master that should happen soon now.

floion, 2018-03-17:
Go it.
Can I ask you to leave a note here when you guys will have a rocko branch?

Thank you

jan-kiszka, 2018-03-17:
We will close this issue then, sure.

floion, 2018-06-22:
Hi. do you also plan to create a distinct `rocko` branch? Branching as such for each Poky release would be of help to users of your BSP layer.
Thank you

jan-kiszka, 2018-06-22:
We currently do not plan to maintain separate branches for multiple versions on our own.

If you identify a difference from master that is needed to work against a still maintained older version of yocto, feel free to request the creation of such a branch based on a PR with those changes.

floion, 2018-06-22:
At some point changes in future Yocto versions will determine you do incompatible changes with older versions. So how is one to know for certain with which Yocto versions is this BSP compatible with?

jan-kiszka, 2018-06-22:
We are very well aware of these dependencies. This is more about how many of them we officially maintain here. The current choice is exactly 1 (linear development).

We document the official, tested versions of all layer dependencies via the kas.yml files. That may not mean that other version combinations do not work, but the committed setup is always the reference (and the foundation of the example image).

When you build your own image, you can refer in your layer kas file to a revision of meta-iot2000 that supports your selection of poky etc.

floion, 2018-06-22:
Okay, thank you for the reply