siemens/meta-iot2000 #70: kas-sdk-linux-x64.yml version not compatible with README.md examples

DaCookie4u, 2018-01-19:
I noticed that the kas command described in README.md for building the SDK is not compatible with the docker build command inside the README.md:

```
docker run -v (pwd):/shared-volume:rw -e USER_ID=(id -u) --rm -t -i \
              kasproject/kas:0.11.0 sh -c "
              cd /shared-volume &&
              git clone https://github.com/siemens/meta-iot2000 &&
              kas build meta-iot2000/meta-iot2000-example/kas-sdk-linux-x64.yml"
```

`kas.includehandler.LoadConfigException: This version of kas is compatible with version 1 to 2, file has version 3: /shared-volume/meta-iot2000/meta-iot2000-example/kas-sdk-linux-x64.yml`

Not sure if that is intended. Changing either the file to version 2, or using the latest kasproject/kas image solves the problem. Although I did not take the time to check whether build successfully completes with the latest docker image. I went for the first solution.

jan-kiszka, 2018-01-19:
The first workaround is not the right on. We need to update the link, likely just removing the specific version reference, suggesting latest. Thanks for the note!

jan-kiszka, 2018-01-25:
Fixed in master.

closed 2018-01-25 by jan-kiszka