siemens/meta-iot2000 #72: armm and upm no longer in intel-iot-middleware

leisipeisi, 2018-01-23:
I just tried to build my own version for the IOT2000 roughly following [this ](https://www.dmcinfo.com/latest-thinking/blog/id/9395/creating-a-linux-image-for-the-siemens-iot2000-series) description (not official, I know). During this, I got into some version incompabatilities and found the following: The newest commits of intel-iot-middleware deleted armm and upm in favour of them being included in openembedded ([source](http://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-iot-middleware/log/)). This caused some "no recipts available for [...]" errors.
Because of this it is necessary to use an older commit of intel-iot-middleware for everything to work (I used commit 821cf14c8304).

I don't know if this is the right place to state this, I just wanted to prevent others to make the same mistake I made :)

jan-kiszka, 2018-01-23:
Have a look at https://github.com/siemens/meta-iot2000/commits/jan/rocko where we already removed that legacy dependency.

Also, don't follow that obsolete description but rather use the new kas-based workflow as the README describes. That will give you a correct and up-to-date image, and that in a much simpler way.

leisipeisi, 2018-01-24:
Thank you!
Could you explain how to add other layers to the kas-based image? That was why I started using the other description.

jan-kiszka, 2018-01-24:
Please have a look at this thread: https://support.industry.siemens.com/tf/ww/en/posts/creating-custom-layer/182600/?page=0&pageSize=10
I've attached an example layer there as well.

closed 2018-01-24 by jan-kiszka