siemens/meta-iot2000 #73: Error creating Example IOT2000 image

nuno407, 2018-01-26:
Hi,

I'm trying to build a Yocto image based in this repository. 
I have followed the README instructions without any success (with and without docker).
I always find the following error:
2018-01-26 20:47:11 - INFO     - NOTE: Running task 1790 of 3415 (/shared-volume/poky/meta/recipes-devtools/gcc/gcc-runtime_6.2.bb:do_compile)
2018-01-26 20:47:11 - INFO     - NOTE: recipe gcc-runtime-6.2.0-r0: task do_compile: Started
2018-01-26 20:54:52 - ERROR    - ERROR: gcc-runtime-6.2.0-r0 do_compile: oe_runmake failed
2018-01-26 20:54:52 - ERROR    - ERROR: gcc-runtime-6.2.0-r0 do_compile: Function failed: do_compile (log file is located at /shared-volume/build/tmp/work/i586-nlp-32-poky-linux/gcc-runtime/6.2.0-r0/temp/log.do_compile.10059

The log file is in the attachments.
How can i correct this issue?

Thank you,
Nuno

jan-kiszka, 2018-01-27:
This is a very weird thing, specifically inside the docker container which works perfectly otherwise.

Does the issue persist when you retry the build without removing the previous results? Call the same docker run for that, just leaving out the git clone.

Could you also try the jan/rocko branch (git clone -b jan/rocko ...)?

nuno407, 2018-01-27:
Hi @jan-kiszka,
Yeap, is a very weird thing.
I actually find the solution in [here](https://stackoverflow.com/questions/36387647/bitbake-fails-at-do-compile-in-gcc-runtime-5-2-0-r0) after some hours of searching.
Happens that because of low space in the linux partition I used a disk in NTFS format to store all the build folder. Seems that this can not be done. 
Please add to the wiki some kind of warning.

Thank you,
Nuno

jan-kiszka, 2018-01-27:
Means the partition the shared-volume was on was NTFS formatted? Is the host a Windows machine?

Feel invited to send a PR that adds some sentence of warning to the README.md.

nuno407, 2018-01-27:
> Means the partition the shared-volume was on was NTFS formatted?
Yes! All the files were in the NTFS formated disk. 

> Is the host a Windows machine?
No, the host is a Linux machine (elementaryOS).

closed 2018-02-18 by jan-kiszka