siemens/meta-iot2000 #76: Unable to build image: Fetcher failure for URL: 'https://www.example.com/'

Loki-Afro, 2018-02-05:
I'm unable to build the bsp image.

with or without docker doesn't matter :/

```
--- Error summary ---
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Fetcher failure for URL: 'https://www.example.com/'. URL https://www.example.com/ doesn't work. Please ensure your network is configured correctly.
```

Why should the build process resolve example.com?

jan-kiszka, 2018-02-05:
It's a check for the connectivity of your build environment. Maybe you are behind some ugly proxy, and you still need to provide the required http_proxy & Co. settings to the build environment?

Loki-Afro, 2018-02-05:
@jan-kiszka maybe using `www.example.com` is not the best choice here?

I am not behind any proxy mess. But my DNS fucked up.

closed 2018-02-05 by Loki-Afro

jan-kiszka, 2018-02-05:
Maybe, but it would be an upstream (OE, Yocto) topic then. We just eat their food here.

LiveFreeAndRoam, 2019-09-23:
I installed a VB VM, following the directions describing how to setup the [OpenBMC development environment](https://gist.github.com/geissonator/71dad3ccca8bd42c02cfb9726df6c171).

I also hit this error.  Basically, my VB VM does not resolve www.example.com, whereas my host PC does.  Anyway, I made the following change and got past this blocker...

In file: `~/openbmc/poky/meta/conf/distro/include/default-distrovars.inc`

Find the line containing:

`CONNECTIVITY_CHECK_URIS ?= "https://www.example.com/"`

Change it to be (or some other URL your VM has access to):

`CONNECTIVITY_CHECK_URIS ?= "https://www.google.com/"`

sachinmokashi, 2020-11-11:
If you want to resolve this issue without modifying `local.conf` or any of the `conf` files for that matter, try doing:

    $touch conf/sanity.conf

It is clearly written in `meta/conf/sanity.conf` that: 

>  Expert users can confirm their sanity with "touch conf/sanity.conf"

If you don't want to execute this command on every session or build, you can comment out the line `INHERIT += "sanity"` from `meta/conf/sanity.conf`, so the file looks something like this:

[![meta/conf/sanity.conf][1]][1]

  [1]: https://i.stack.imgur.com/mNYlL.png

LiveFreeAndRoam, 2021-11-26:
I recently came back to this issue with a fresh clone of OpenBMC.  I had time to learn more about the cause of this problem and found the solution.

At the heart of this issue, it is a certificate problem on my VM.  Basically, when my laptop is connected to the corporate VPN all https traffic goes through the corporate proxy which (understandably) uses its own certificate, effectively becoming a MITM.  Running commands like `curl https://www.google.com` would fail because they did not recognize the corporate certificate.

> As an aside, back in 2019 my access to https://www.google.com succeeded, whereas access to https://www.example.com would fail. Now they both fail, as did many other https accessible websites.

The exact error was:
```
$ curl https://www.google.com
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
```

So the challenge was to add the corporate MITM certificate to the trusted certifcates on my Ubuntu VM.  I followed these steps:

## Identify the MITM certificates
1. Using Firefox, I accessed the https://www.google.com
2. Firefox displayed a warning page, indicating a CA issue.  Clicking `Advanced...` reveals more info with an
`Error code: MOZILLA_PKIX_ERROR_MITM_DETECTED`
3. Select `View Certificate`
4. Notice there will be multiple tabs (one for www.google.com, and others are the certificates you care about)
5. Select the 2nd tab (it will have a name related to the Root CA that needs to be trusted)
6. Scroll down to download `PEM (cert)` and save it (the filename could not be changed). It was saved as `www-google-com.pem`
7. Rename that file to be something like `mitm-1.pem`
8. For each subsequent tab, repeat steps 6-7, naming each file uniquely

At this point you will have some number of files `mitm-*.pem`. The next step is to install these certificates onto your guest OS. My guest OS is Ubuntu 20.04, so I'll provide the steps for that...

## Install the MITM Certificates on Ubuntu
1. Rename each of the downloaded `mitm-*.pem` files to `mitm-*.crt` (you don't convert them, just rename them)
2. Move `mitm-*.crt` to `/usr/local/share/ca-certificates` (reportedly, they need the `.crt` suffix)
3. Now install them with: `sudo update-ca-certificates --fresh`
4. Retry `curl https://www.google.com`, at this point it worked for me.

From here on, I could happily run bitbake and the CONNECTIVITY_CHECK_URIS default setting no longer caused problems.

FWIW, the method OpenBMC uses to validate connectivity is probably wrong because I could set `CONNECTIVITY_CHECK_URIS=""` and the build would succeed, since git access always worked fine. The check should probably be changed to mimic the requirements of the build.