siemens/meta-iot2000 #78: ERROR: linux-libc-headers-4.4-r0 do_fetch: Fetcher failure

deinok, 2018-02-27:
Seems like it fails when it try to fetch git://git.kernel.org/pub/scm/linux/kernel/git/bwh/linux-cip;protocol=https

Attach of log of the `kas` command
[build.log](https://github.com/siemens/meta-iot2000/files/1762687/build.log)

```
ERROR: linux-libc-headers-4.4-r0 do_fetch: Fetcher failure: Fetch command export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/deinok/meta-iot2000/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/deinok/meta-iot2000/poky/scripts:/home/deinok/meta-iot2000/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux:/home/deinok/meta-iot2000/build/tmp/sysroots/iot2000/usr/bin/crossscripts:/home/deinok/meta-iot2000/build/tmp/sysroots/x86_64-linux/usr/sbin:/home/deinok/meta-iot2000/build/tmp/sysroots/x86_64-linux/usr/bin:/home/deinok/meta-iot2000/build/tmp/sysroots/x86_64-linux/sbin:/home/deinok/meta-iot2000/build/tmp/sysroots/x86_64-linux/bin:/home/deinok/meta-iot2000/poky/scripts:/home/deinok/meta-iot2000/poky/bitbake/bin:/usr/sbin:/usr/bin:/sbin:/bin"; export HOME="/tmp/tmp522xqn_p"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror https://git.kernel.org/pub/scm/linux/kernel/git/bwh/linux-cip /home/deinok/meta-iot2000/build/downloads/git2/git.kernel.org.pub.scm.linux.kernel.git.bwh.linux-cip --progress failed with exit code 128, output:
Cloning into bare repository '/home/deinok/meta-iot2000/build/downloads/git2/git.kernel.org.pub.scm.linux.kernel.git.bwh.linux-cip'...
```

BaochengSu, 2018-02-28:
How much memory do you get for your build machine?

jan-kiszka, 2018-02-28:
Yeah, there is some out of memory error reported, besides that git fetch failure. The fetch is terminated via SIGKILL, and that might have been the out-of-memory killer of Linux. You should see this in your system log.

deinok, 2018-02-28:
Thanks @BaochengSu and @jan-kiszka I will try to add about 8GB of memory.
I will report back ASAP.

deinok, 2018-03-01:
Fixed, thanks for the help.

closed 2018-03-01 by deinok

Saicharan2001, 2023-02-24:
> 

No I am unable to fetch it all alone

jan-kiszka, 2023-02-24:
Use 'master' instead. Or patch the url to git.kernel.org/pub/scm/linux/kernel/git/cip/linux-cip.git.