siemens/meta-iot2000 #79: Websever using Apache, SQL and PHP

leisipeisi, 2018-03-05:
Hi everyone,

since I had some hard time creating an image for IOT2000 using apache, SQL and PHP, I wanted to share my solution with others to save them some time. I am not sure if this is the right place to do so. If there is a better place, I would appreciate if you would transfer the information to there.

As a base, a used the example layer of this thread: [Link](https://support.industry.siemens.com/tf/ww/en/posts/creating-custom-layer/182600/?page=0&pageSize=10)

Add the following to `meta-iot2000-example/kas.yml`:
```
meta-openembedded:
	layers:
		meta-webserver:
		meta-oe:
		meta-networking:
		meta-python:
```

Create a the new file (if not existing) `meta-iot2000-example/receipes-code/images/iot2000-example-image.bbappend` and add the following content:
```
DEPENDS= "apache2 uclibc"
RDEPENDS_${PN} = "apache2 uclibc"
PACKAGECONFIG_append += " mysql apache2 ${@bb.utils.contains('DISTRO_FEATURES', 'pam, 'pam', '', d)}"
 
IMAGE_INSTALL_append += " phpmyadmin"
IMAGE_INSTALL_append += " proftpd"
IMAGE_INSTALL_append += " apache2"
IMAGE_INSTALL_append += " mysql5"	
IMAGE_INSTALL_append += " php"
IMAGE_INSTALL_append += " php-cli"
IMAGE_INSTALL_append += " php-modphp"
```

Add the following to meta-iot2000-example/conf/layer.conf:
```
PACKAGECONFIG_append_pn-php= " apache2"
PACKAGECONFIG_append_pn-php = " uclib"
```
Make sure to not forget the spaces between " and the beginning of the package name.

Compile everything with `kas build meta-iot2000-example/kas.yml` and write the resulting `.wic` file to the IOT2000.

To make apache and php work together, add the following lines to `/etc/apache2/httpd.conf` on the IOT2000:
```
LoadModule php5_module lib/apache2/modules/libphp5.so

<FilesMatch \.php$>
	SetHandler application/x-httpd-php
</FilesMatch>
```
Then you can restart apache using `/etc/init.d/apache2 restart`

I hope this helps somebody!

jan-kiszka, 2018-03-06:
Thanks for sharing! Probably a better place is the IOT2000 user forum.

The DEPENDS and RDEPENDS of the image recipe on apache2 shouldn't be required. Adding that package to IMAGE_INSTALL does that for you.

Also, there should be no reason to add uclibc. We have both glibc (the main libc, also used by php) and uclibc on board, but the latter is only used by the Arduino runtime. Yocto is confused due to the presence of both and generates a couple of harmless warnings. If those warnings all disappeared with your changes, we may have to look into what they actually did.

leisipeisi, 2018-03-07:
You are right, the inclusion of uclibc indeed helped solving these warnings. If I can help you with any additional information on this, please let me know!

jan-kiszka, 2018-03-07:
Cool. If you can explain why that is needed and write a patch for current master, we would be happy to merge that!

leisipeisi, 2018-03-07:
Saying "information" I meant something like the warnings I received or similar:D 
I have absolutely no clue on how to fix it, sorry.

jan-kiszka, 2018-08-14:
FWIW, those uclibc-related warnings have been resolved upstream meanwhile.

closed 2018-08-14 by jan-kiszka