siemens/meta-iot2000 #82: Add latest version of boost library

nuno407, 2018-04-22:
Hi,
According to bergmanu in [this post](https://support.industry.siemens.com/tf/ww/en/posts/modifying-the-sd-card-image-adding-new-layers/177554?page=0&pageSize=10#pagetop) is necessary to add a layer that contain the recipe for boost. As far as I can understand the necessary recipe is [this](http://layers.openembedded.org/layerindex/recipe/5268/) and is located in the openembedded-core layer. Cloning this layer and adding it to the bblayers.conf ends in bitbake error.

```
ERROR: Unable to parse /media/nuno/yocto_build/meta-iot2000/poky/bitbake/lib/bb/data_smart.py
Traceback (most recent call last):
  File "/media/nuno/yocto_build/meta-iot2000/poky/bitbake/lib/bb/data_smart.py", line 400, in DataSmart.expandWithRefs(s="${@os.path.dirname(bb.utils.which(d.getVar('PATH'),'bitbake'))}:${HOSTTOOLS_DIR}", varname='PATH[:=]'):
                 except Exception as exc:
    >                raise ExpansionError(varname, s, exc) from exc
     
bb.data_smart.ExpansionError: Failure expanding variable PATH[:=], expression was ${@os.path.dirname(bb.utils.which(d.getVar('PATH'),'bitbake'))}:${HOSTTOOLS_DIR} which triggered exception TypeError: getVar() missing 1 required positional argument: 'expand'

```

How can I add the 1.66 boost version to the image/build ?
Is there any tool to, using a GUI, select the desired recipes to be included in the build?

Thank you, 
Nuno

jan-kiszka, 2018-04-24:
There is support for boost 1.66 in poky master, i.e. in no release yet. However, there is jan/rocko to get you to yocto 2.4 which comes with 1.64. From that on, you may add 1.66 by pulling the upstream changes in your own layer that provides a 1.66 boost recipe.

nuno407, 2018-04-24:
I'm currently using the master branch and the installed version is the 1.61. can you please indicate the easiest way to create a custom layer containing boost library?
Thank you, Nuno

jan-kiszka, 2018-04-26:
There is is an example customization layer available here: https://support.industry.siemens.com/tf/ww/en/posts/creating-custom-layer/182600/?page=0&pageSize=10. But actually implementing the v1.66 recipe (bbappend for existing recipe, or a new that includes the common boost recipe artifacts) can be more tricky.

nuno407, 2018-04-27:
Hi @jan-kiszka,

I already have successfully created a customization layer but the used boost package was v1.61. 
How can I force the use of a specific version?

Thank you,
Nuno

jan-kiszka, 2018-08-14:
Oh, missed the follow-up question back then. Does it still apply?

nuno407, 2018-08-23:
Hi @jan-kiszka ,
Thank you but I find a way to not use that version.
Please close issue

closed 2018-08-23 by jan-kiszka