siemens/meta-iot2000 #85: Default root password is not empty

garlix, 2018-05-28:
Hi,
just start playing with the IOT2040. Firstly, I flashed an SD card with the prebuilt `example-v2.2.0.wic` found online. Then, I git-cloned this repo and build my own image and added the meta-qt5 framework. The problem is that the output image, built with `kas`, has a non-empty root password. Why? Which password is it? How can I change it?
My actual workaround was removing the root password from the `/etc/shadow` file

jan-kiszka, 2018-05-28:
I strongly suspect this behavioral change came from the customizations. Our vanilla example & bsp images come with an empty root password. And that's due to

    debug-tweaks: |
      EXTRA_IMAGE_FEATURES = "debug-tweaks"

in meta-iot2000-bsp/kas.yml.

garlix, 2018-05-29:
I've checked out your hint and you were right: I accidentally overwrote `EXTRA_IMAGE_FEATURES` in `kas.yml` file, instead of just appending to it.

OT:
I've added `meta-qt5` layer to `iot2000-example-image` recipe. Everything built successfully but when I run a Qt application on the IOT2000 it crashes with a SIGILL and an `Illegal instruction` message. If I do a backtrace, the issue seems to be in the `qCalculateBlockSize` of Qt core lib. Do I have to take some particular expedient when building meta-qt5? I've added it in the `kas.yml` file with
```
  meta-qt5:
    url: https://github.com/meta-qt5/meta-qt5
    refspec: 73f99f2370a3dc5b81f42d0af7fe530431599454
```
under the `repos` section

jan-kiszka, 2018-05-29:
Most likely, you are building Qt with some compiler optimization flags or even optimized assembly that does not run on the Quark. It's a Pentium-I device, thus no SSE & Co. instructions that tend to be used for accelerating things, even memcpy. Check if you can configure Qt without that or for a specific processor generation. That may require overloading the Qt build recipe then.

garlix, 2018-05-30:
Thank a million @jan-kiszka ! You saved my days! You were totally right, I've disabled sse2 optimizations (enabled by default starting from QT 5.3) and now it seems to work. Do you think could worth something if I do a pull request with the modifications I've made in order to have QT working?

jan-kiszka, 2018-05-30:
Yes, that would be valuable! We should make this an optional fixup, though: Such a bbappend would become active if a custom image is built that includes the Qt meta layer, but we would not make that layer a hard dependency (there are currently no plans to include Qt into the example image).

Look at how we handle Go in the BSP layer for that pattern. It basically consists of having a dummy target for the .bbappend in case the user does not include a layer with the target recipe.

closed 2018-05-30 by jan-kiszka