siemens/meta-iot2000 #88: Question - MPCIE

deinok, 2018-06-28:
Hi, I'm trying to use a Telit HE910-G.
It's just a 3g mPCIe.

I was expecting to be "plug and play" but seems like the SO is not generating an ethernet adapter,

I'm missing something?

jan-kiszka, 2018-06-29:
Are you sure the modem exposes itself as an Ethernet interface? Maybe it just adds another uart and expects a PPP service to issue AT commands over that.

Are you using the latest master revision with oFono support? Did you check if iot2000setup detected it? Also check the system logs for hints if the modem is detected.

deinok, 2018-06-29:
-  Seems like by default it dont expose itself as an ethernet interface.
- I'm using meta-iot2000-example 2.2
- Command `lsusb` dont seems to detect the mpcie.
- Also, seems like no ppp, pppd or ofono executables are instalated (Nothing related in $PATH or in /etc)
- iot2000setup -> I dont see anything related to its detection

Is there any "flag" or something i should enable to access PPP support?

jan-kiszka, 2018-06-29:
For ofono/connman support (obsoletes pppd), you need to build an own image from current master. 2.2 does not contain that feature yet.

You should see the card via "lspci -knn". That may also list some driver that picked it up. If there is none listed, please share the PCI ID.

deinok, 2018-06-30:
```log
root@iot2000:~# lspci -knn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:0958]
        Subsystem: Intel Corporation Device [8086:095e]
        Kernel driver in use: iosf_mbi_pci
00:14.0 SD Host controller [0805]: Intel Corporation Device [8086:08a7] (rev 10)
        Subsystem: Intel Corporation Device [8086:08a7]
        Kernel driver in use: sdhci-pci
00:14.1 Serial controller [0700]: Intel Corporation Device [8086:0936] (rev 10)
        Subsystem: Intel Corporation Device [8086:0936]
        Kernel driver in use: serial
00:14.2 USB controller [0c03]: Intel Corporation Device [8086:0939] (rev 10)
        Subsystem: Intel Corporation Device [8086:0939]
        Kernel driver in use: pch_udc
00:14.3 USB controller [0c03]: Intel Corporation Device [8086:0939] (rev 10)
        Subsystem: Intel Corporation Device [8086:0939]
        Kernel driver in use: ehci-pci
00:14.4 USB controller [0c03]: Intel Corporation Device [8086:093a] (rev 10)
        Subsystem: Intel Corporation Device [8086:093a]
        Kernel driver in use: ohci-pci
00:14.5 Serial controller [0700]: Intel Corporation Device [8086:0936] (rev 10)
        Subsystem: Intel Corporation Device [8086:0936]
        Kernel driver in use: serial
00:14.6 Ethernet controller [0200]: Intel Corporation Device [8086:0937] (rev 10)
        Subsystem: Intel Corporation Device [8086:0937]
        Kernel driver in use: stmmaceth
00:14.7 Ethernet controller [0200]: Intel Corporation Device [8086:0937] (rev 10)
        Subsystem: Intel Corporation Device [8086:0937]
        Kernel driver in use: stmmaceth
00:15.0 Serial bus controller [0c80]: Intel Corporation Device [8086:0935] (rev 10)
        Subsystem: Intel Corporation Device [8086:0935]
        Kernel driver in use: pxa2xx_spi_pci
00:15.1 Serial bus controller [0c80]: Intel Corporation Device [8086:0935] (rev 10)
        Subsystem: Intel Corporation Device [8086:0935]
        Kernel driver in use: pxa2xx_spi_pci
00:15.2 Serial bus controller [0c80]: Intel Corporation Device [8086:0934] (rev 10)
        Subsystem: Intel Corporation Device [8086:0934]
        Kernel driver in use: intel_quark_mfd_i2c_gpio
00:17.0 PCI bridge [0604]: Intel Corporation Device [8086:11c3]
        Kernel driver in use: pcieport
00:17.1 PCI bridge [0604]: Intel Corporation Device [8086:11c4]
        Kernel driver in use: pcieport
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:095e]
        Subsystem: Intel Corporation Device [8086:095e]
        Kernel driver in use: lpc_sch
02:00.0 Serial controller [0700]: Exar Corp. Device [13a8:0352] (rev 03)
        Kernel driver in use: exar_serial
```

deinok, 2018-06-30:
Hey, seems like iotsetup2000 is detecting the modem

jan-kiszka, 2018-07-01:
Hmm, according to your lspci, there is no mPCIe device. Are you sure the adapter is fine and properly plugged?

deinok, 2018-07-01:
Yes, using this ->https://embexus.com/tag/ofono/ (The ofono setup section)

- ./enable-modem -> Connecting modem /telit_0... - Seems good
- ./create-internet-context telefonica.es telefonica telefonica - Seems good
- ./online-modem ->Setting modem /telit_0 online... - Seems good
- ./activate-context -> Error activating /telit_0/context1: org.ofono.Error.NotAttached: GPRS is not attached - Seems the error

deinok, 2018-07-03:
@jan-kiszka  I fixed it. The problem was with my ISP provider and roaming.

So, for anybody trying to give 3G to the IOT, iot2000setup works pretty well. Also, the commands I have descrived can be used to do it without iot2000setup.

closed 2018-07-03 by deinok

deinok, 2018-07-04:
Type: mPCIe
Manufacturer: Telit
Model: HE910-D
Country: Spain
Notes: It requires to add two 3G antenas in the two connectors of the mPCIe

Notable Issues found: None related to IOT2000 SO or Hardware

vitus1984, 2018-09-20:
Dear deinok, 
in my case it is not working out following your instructions related to the guide here: https://embexus.com/tag/ofono/
The problem is already that trying "./list-modems" gives me following as output:

[ /telit_0 ]
    Powered = 0
    Online = 0
    Features =
    Lockdown = 0
    Interfaces =
    Type = hardware
    Emergency = 0

At the same time it is possible to use AT commands under ttyACM0 in a proper way but to establish a ppp0 via pppd procedure is just not working out :-(

deinok, 2018-09-20:
Can you show an image of the pcie hardware?
What is your country?
Commit of the image that you are using?

vitus1984, 2018-09-21:
- I tried both mPCIe HE910- , with GPS and w/o GPS 
[Screenshots and photo  related to problem.pdf](https://github.com/siemens/meta-iot2000/files/2404537/Screenshots.related.to.problem.pdf)
- Germany
- the latest official master (dee5ab7)

vitus1984, 2018-09-21:
Maybe I have to check the kernel configuration and add some stuff here?

deinok, 2018-09-21:
./enable-modem -> Connecting modem /telit_0... - Seems good
./create-internet-context telefonica.es telefonica telefonica - Seems good
./online-modem ->Setting modem /telit_0 online... - Seems good
./activate-context -> Error activating /telit_0/context1: org.ofono.Error.NotAttached: GPRS is not attached - Seems the error

> Yes, using this ->https://embexus.com/tag/ofono/ (The ofono setup section)
> 
> * ./enable-modem -> Connecting modem /telit_0... - Seems good
> * ./create-internet-context telefonica.es telefonica telefonica - Seems good
> * ./online-modem ->Setting modem /telit_0 online... - Seems good
> * ./activate-context -> Error activating /telit_0/context1: org.ofono.Error.NotAttached: GPRS is not attached - Seems the error
> 
> ```
> ./list-modems
> [ /telit_0 ]
>     Emergency = 0
>     Manufacturer = Telit
>     Lockdown = 0
>     Model = HE910-D
>     Powered = 1
>     Revision = 12.00.026
>     Interfaces = org.ofono.NetworkRegistration org.ofono.ConnectionManager org.ofono.AllowedAccessPoints org.ofono.SimManager
>     Features = net gprs sim
>     Online = 1
>     Serial = 359180083086675
>     Type = hardware
>     [ org.ofono.NetworkRegistration ]
>         Name =
>         Status = searching
>         Mode = auto
>     [ org.ofono.ConnectionManager ]
>         Bearer = lte
>         RoamingAllowed = 0
>         Attached = 0
>         Powered = 1
>     [ org.ofono.AllowedAccessPoints ]
>     [ org.ofono.SimManager ]
>         ServiceNumbers = [Recarga Saldo] = '2200' [Buzon Movistar] = '123' [Inf. 11822] = '11822' [ProgramaPuntos] = '2236' [ConsultConsumo] = '2266' [Noticias] = '22303' [Yavoy] = '2210' [At. Cliente] = '1004'
>         MobileCountryCode = 214
>         Present = 1
>         PinRequired = none
>         SubscriberNumbers =
>         SubscriberIdentity = 214075526485192
>         BarredDialing = 0
>         LockedPins =
>         Retries = [puk = 10] [pin = 3] [puk2 = 10] [pin2 = 3]
>         CardIdentifier = 8934072700023821515
>         MobileNetworkCode = 07
>         FixedDialing = 0
>         PreferredLanguages = es
> ```

Can you exec the following commands and show us the CLI results?

vitus1984, 2018-09-21:
I have already did above I think...: 

after executing the script ./list-modems under /usr/lib/ofono/test

[ /telit_0 ]
Powered = 0
Online = 0
Features =
Lockdown = 0
Interfaces =
Type = hardware
Emergency = 0

deinok, 2018-09-21:
okey, have you tryed in this order?
./enable-modem -> Connecting modem /telit_0... - Seems good
./create-internet-context telefonica.es telefonica telefonica - Seems good
./online-modem ->Setting modem /telit_0 online... - Seems good
./activate-context -> Something I dont remember

vitus1984, 2018-09-21:
Yes. Unfortunately in case of "./enable-modem" error messages would come up: 

root@iot2000:/usr/lib/ofono/test# ./enable-modem
Connecting modem /telit_0...
Traceback (most recent call last):
  File "./enable-modem", line 20, in <module>
    modem.SetProperty("Powered", dbus.Boolean(1), timeout = 120)
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.ofono.Error.Failed: Operation failed

I am wondering why "./list-modem" brings "Powered=0" , "Online=0" etc. although it is possible to use the modem properly if using screen /dev/ttyACM0 and sending AT commands manually...
 
I also build an image today night where I have changed the Kernel configuration on basis of the information given at oFono - Embexus 
under recipes-kernel/linux/configs/iot2000example.cfg ...maybe I am doing sth. wrong here?

deinok, 2018-09-21:
Seems like the Trackback is not giving so much information.
In my case it works well without any change to the image, this commit works well with me (https://github.com/siemens/meta-iot2000/commit/dee5ab773d30a20034d849ab830c2abb893c799c).
Can you connect the mPCIE to a Desktop with a Debian based SO? (Debian or Ubuntu should work well for the tests)

vitus1984, 2018-09-21:
But have you loaded specific Kernel modules according to the USB manual from TELIT maybe?
[image]

I even have changed from 12V to 24V adapter to make sure that is nothing having to do with the hardware although like I said before it is working fine with AT command also at present 12V

I will report after having tried some more Kernel adjustments

deinok, 2018-09-21:
I'm powering IOT2000 with 24V/0,6A with no kernel modules modifications

vitus1984, 2018-09-21:
Is the mPCIe HE910-D exposed as ttyACMx or ttyUSBx in your case under /dev?

deinok, 2018-09-21:
[image]
Can you check that the Model is the same? And works with a SIM?

deinok, 2018-09-21:
ttyACMx

vitus1984, 2018-09-21:
[image]
...I have both (w and w/o GPS option) working with a SIM ...at the moment the HE910 is inside ... I will try now again with the other one...

vitus1984, 2018-09-21:
Oh man. It is the hardware ...I messed up before ...: 
It is working with the HE910-D (without GPC) but not with HE910 (with GPS)... Finally. 
Sorry for the trouble... and BIG THANKS to you for helping out deinok!!!
At least now I can proceed with testings ...but its sad that it is not with the HE910 :-(

deinok, 2018-09-21:
Dont worry we are here to help each other, I have not tested the other model you have, I'm sorry to be unable to help :(

vitus1984, 2018-09-21:
Everything working out now but as also in your case the fourth command: "./activate-context" led to an error message although "./list-modems" in my case indicates following which I interprete as positive. 
What was the solution in your case for this?

root@iot2000:/usr/lib/ofono/test# ./list-modems
[ /telit_0 ]
    Revision = 12.00.024
    Online = 1
    Lockdown = 0
    Serial = 359180081355254
    Interfaces = org.ofono.ConnectionManager org.ofono.NetworkRegistration org.ofono.AllowedAccessPoints org.ofono.SmartMessaging org.ofono.PushNotification org.ofono.MessageManager org.ofono.Phonebook org.ofono.SimManager
    Type = hardware
    Manufacturer = Telit
    Features = gprs net sms sim
    Emergency = 0
    Powered = 1
    Model = HE910-D
    [ org.ofono.ConnectionManager ]
        RoamingAllowed = 0
        Bearer = hsdpa
        Powered = 1
        Attached = 1
        Suspended = 0
    [ org.ofono.NetworkRegistration ]
        Technology = umts
        LocationAreaCode = 955
        MobileNetworkCode = 02
        Strength = 60
        CellId = 246357245
        MobileCountryCode = 262
        Status = registered
        Name = M2M
        Mode = auto
    [ org.ofono.AllowedAccessPoints ]
    [ org.ofono.SmartMessaging ]
    [ org.ofono.PushNotification ]
    [ org.ofono.MessageManager ]
        ServiceCenterAddress = +491722270333
        Bearer = cs-preferred
        UseDeliveryReports = 0
        Alphabet = default
    [ org.ofono.Phonebook ]
    [ org.ofono.SimManager ]
        FixedDialing = 0
        Retries = [pin = 3] [puk = 10] [pin2 = 3] [puk2 = 10]
        MobileNetworkCode = 02
        ServiceProviderName = M2M
        PinRequired = none
        BarredDialing = 0
        MobileCountryCode = 262
        Present = 1
        LockedPins = pin
        SubscriberNumbers =
        SubscriberIdentity = 262021804673647

deinok, 2018-09-21:
`./activate-context -> Error activating /telit_0/context1: org.ofono.Error.NotAttached: GPRS is not attached - Seems the error` In my case the solution was connecting the hardware antennas...

vitus1984, 2018-09-24:
Well in my case it can not be the antennas as the AT commands and registering to provider via classical way with AT commands is no problem. Still the "./activate-context" command is resulting in "org.ofono.Error.Failed: Operation failed"" error message.
I also have no clue at which point the script throws an exception. Is it possible to find it out somehow and to track it a little bit more?

yizenlim, 2020-07-22:
glad i found this post, @vitus1984 , does that mean the HE910-G will not work with the iot2000 ? , is it a must to connect both antennas on the Telit mPcie for it to work ?

vitus1984, 2020-07-22:
I think it can work. But I havent managed to realize it although with the one model w/o GPS it was possible and successful. 
You would need to connect both antennas.