siemens/meta-iot2000 #89: mraa i2c read IOError Connection timed out

mrlowalowa, 2018-07-05:
Hello,

I wanted to read 45 Bytes from a I²C device. When I do so I get the following error:

```
ssh root@10.42.0.114
root@iot2000:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- 1d -- -- 
20: -- -- -- -- -- UU UU UU 28 -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- -- 
50: -- -- -- -- 54 55 56 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
root@iot2000:~# cd gits/test_python_bno055/      
root@iot2000:~/gits/test_python_bno055# python publish_imu_roslibpy.py 
Traceback (most recent call last):
  File "publish_imu_roslibpy.py", line 82, in <module>
    ros_client.on_ready(publish_imu_data(publisher))
  File "publish_imu_roslibpy.py", line 56, in publish_imu_data
    buf = bno.readBytes(bno055.BNO055.BNO055_ACCEL_DATA_X_LSB_ADDR, 45)
  File "/home/root/gits/test_python_bno055/bno055.py", line 313, in readBytes
    return self._i2c.readBytesReg(register, numBytes)
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1198, in readBytesReg
    return _mraa.I2c_readBytesReg(self, reg, data)
IOError: [Errno 110] Connection timed out
root@iot2000:~/gits/test_python_bno055# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- UU UU UU -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

Is this an already known issue?
The I²C line is broken after this call. After a reboot of the IoT2040 everything works again.
The error is repeatable.

jan-kiszka, 2018-07-05:
Sounds a bit like the targeted device (0x1d or 0x28?) is getting angry and blocks the bus afterwards.

You can test if i2c communication works via mraa's python binding in general by talking to that AT24 EEPROM at 0x54.

jan-kiszka, 2018-08-14:
Is this issue still persisting?

mrlowalowa, 2018-08-14:
Hello,

Reading 300 bytes from 0x54 is no problem. Seems to be related to the BNO055 device that I’m using.

Thank you for your help! :)

closed 2018-08-14 by mrlowalowa

HelloHungryImDad, 2018-08-25:
Sorry to necropost, but I am facing the same issue with a BMP280 sensor.

I can read a single byte from a register without issue but, like mrlowalowa, multiple byte reads causes the bus to block and needs a reboot to fix. I suspect this may be due to an inappropriate I2C bus speed for these sensors. 

I verified that the same sensor works on Arduino and Raspberry Pi, whose I2C buses default to standard mode (100 kbits/s), whereas the IoT2000's bus defaults to fast mode (400 kbits/s). On the IoT2020 If instead of using the mraa function for reading multiple bytes, I instead call the single byte read multiple times I get all the data back and the bus stays up. Since this method is slower than the multiple byte in one read method, it follows that the bus speed is too high for this sensor.

Is there any way to adjust the I2C bus speed on the IoT2020? I checked the [mraa documentation for galileo](https://iotdk.intel.com/docs/master/mraa/galileorevh.html) but the IoT2020 does not appear to be using the same driver for the I2C chip.

I also have a thread running on the [Siemens IoT2000 forum](https://support.industry.siemens.com/tf/ww/en/posts/bmp280-sensor-and-iot2020/197037/?page=0&pageSize=10) with additional test results.