siemens/meta-iot2000 #91: kas build doesn't finish doing a task

mikelga, 2018-07-18:
Hello,

I'm trying to create an image for the sd following the page https://github.com/siemens/meta-iot2000. For that, I'm using Debian operative system in a virtual machine. When I do **kas build meta-iot2000-example/kas.yml**, it start to do the tasks correctly, but it stays frozen in a task for many hours. This information appears in the task: 
**INFO - NOTE: Running task 85 of 4302 (/root/meta-iot2000/poky/meta/recipes-devtools/binutils/binutils-cross_2.29.1.bb:do_fetch)
INFO - NOTE: recipe binutils-cross-i586-2.29.1-r0: task do_fetch: Started**
I have used two different virtual machines (one with Debian 8 and the other with Debian 9) and in both machines it has been frozen in the same task. What is this about?

Thanks,

jan-kiszka, 2018-07-18:
Whenever fetching steps are not progressing, that may point to proxy or general networking issues. OTOH, those activities tend to time out eventually. Does the container have network connection? Do other fetch steps before this one succeed?

mikelga, 2018-07-18:
Hi, thanks for the quick reply!
Yes, the Debian has a network connection. I am building the image using Native Build, and not Docker Build. The other fetch steps does correctly.
[image]

jan-kiszka, 2018-07-18:
Can you identify in the process list which task actually hangs (ps aux, pstree)?

mikelga, 2018-07-18:
Hi,

Sorry, but I don't understand your question very well. The task that remains hanging is "recipe binutils-cross-i586-2.29.1-r0". 

I have another doubt: how long does it last, more or less, the execution of **kas build meta-iot2000-example/kas.yml**

Thanks,

mikelga, 2018-07-20:
Hello,

Finally, it finished doing all the tasks after more than 30 hours. For that, I used Debian 8 Jessie operative system in the VirtualBox program. Then, I inserted the sd card in the device IoT2040 and it works correctly.

I want to run a program with python language on the device, and this program uses several packages. To install these packages, I used the following commands in the Debian:

- apt-get install python-dev

- apt-get install python-numpy

- pip install psutil

- pip install apiclient

- pip install python-snap7

- pip install opcua

- ...

But, after passing the image to the IoT2040 device, it does not let me use those packages installed in the Debian.

Besides, it does not let me install the packages from the IoT2040 because it does not recognize the apt-get and pip commands. 

What steps should I follow to run my python program on IoT2040 using the opcua, python-dev... packages?

Thanks,

jan-kiszka, 2018-07-20:
30h - well, that's unfortunately the price of using a VirtualBox VM, likely with limited host resources as well. On a proper server, this can be done in 45 minutes.

Regarding the packages: Yocto does not magically transfer packages you install on the build host into the target image. What you rather need are changes like in 7c3ef9d5c6becca9b41c026f2db1ff6c9bd5c955.

mikelga, 2018-07-27:
Hi, 

I want to install the grpcio package in the image of Yocto. For this, I installed it first with "pip install grpcio". The installation went correctly. Then, I edited the file **iot2000-example-image.inc**:
**IMAGE_INSTALL_append = " six futures enum34 grpcio"**

When doing **kas build meta-iot-example/kas.yml**, I see this error:
**ERROR: Nothing RPROVIDES 'grpcio' (but /root/meta-iot2000/meta-iot2000-example/recipes-core/images/iot2000-example-image.bb RDEPENDS on or otherwise requires it).**

Thanks,

nospam2000, 2018-07-29:
please have a look at [the openembedded layer/recipe index](http://layers.openembedded.org/layerindex/branch/master/recipes/?q=grpc).
You will see that there is no yocto recipe with the name "grpcio", but several others.
For python you probably want python-grpcio and maybe python-grpcio-tools.

In the column you also see the layer where the recipe is defined. When you don't already have that layers, you need to download and add it.

mikelga, 2018-07-30:
Hi,
Sorry, I am new in Yocto and I don't know the concrete steps that I must follow.
What concrete steps should I follow to be able to use python-grpcio?
Thank you!

nospam2000, 2018-07-30:
try 
`IMAGE_INSTALL_append = " six futures enum34 python-grpcio"`

mikelga, 2018-08-01:
Hi!
I tried doing **IMAGE_INSTALL_append = " python-grpcio"** and it worked correctly
Thank you!

closed 2018-08-14 by jan-kiszka