siemens/meta-iot2000 #92: integrating meta-python layer and library dependencies

laura-mc, 2018-07-18:
Some python packages need "meta-python" layer to be included in the base image (e.g.: python-lxml). It would be grateful to have this layer in the example image, not only because of this package, but also because of many other python functionalities. In the same way, I think there are some useful packages to include in the base image, such as: libxml2, libxslt, zlib, python3-pip, python3-setuptools, python3-modules...

klausnuredini, 2018-07-18:
A few other libs/tools that would be useful to have, if possible, are: 
- PyPy for Python2.7
- easy_install -- part of the python setuptools (Python2.7 also)

jan-kiszka, 2018-07-19:
OK, can be added.

But I would refrain from adding those dependencies in legacy python-2.7 form anymore. Yocto moved on to python 3, and those 2.7 packages may break or get removed. Any concerns about that constraint?

jan-kiszka, 2018-07-19:
BTW, python3-pip and python3-modules are already part of our image. Adding python3-setuptools and python3-lxml now, then checking what the latter already pulls in automatically.

Unfortunately, PyPy looks much more complex on second sight. I would recommend building it manually using our cross-toolchain.

klausnuredini, 2018-07-20:
I managed to install the libs that I was looking for without needing the easy_install tool, so on my side, I wouldn't need any new additional package on the image.

laura-mc, 2018-07-24:
@jan-kiszka I think it is complete, I would send it.

jan-kiszka, 2018-08-14:
merged into master

closed 2018-08-14 by jan-kiszka

mrlowalowa, 2018-08-20:
I would like to reopen this issue as I'm unable to install pip2.
Python2.7 is/was strongly needed as the MRAA Library is/was only available in Python2.7
(since when is MRAA available for Python3?? My build from 06.May.2018 does not contain it? But my new Image from 15. August.2018 contains MRAA for Python3)

I solved this issue back then by loading a few necessary libraries first from the opkg sources and two missing ones directly from  github.com/python/cpython/lib

With the new image I hoped to transfer the missing libraries from the old to the new image.
But this is not working flawless as I need zlib and some other core libs like xml to be installed properly.

Hm. Now I am a bit uncertain what to do. I think the best would be to transfer my code to Python3.

Nevertheless can someone say me if MRAA was recently added as a Python3 library in the example image or was this just a missing configuration from my site?

Cheers

mrlowalowa, 2018-08-20:
Thank you for the fast reply. Python3 should be no issue for me. It was just a pita to make my whole system work with Python2.7 just to find out that MRAA also works with Python3 now.

So MRAA happend to merge into master inbetween my two builds is this right?
Sorry for the confusion in my comment before. I was just uncertain what to do now finding the Python3 integration halfway writing this comment as I was checking the facts.. :laughing:

Just want to confirm this integration and I'm happy with it.. :sake:

jan-kiszka, 2018-08-20:
MRAA is not part of meta-iot2000. We rather pull its recipes from meta-openembedded now (meta-intel-iot-middleware, the former source, is dead by now).

mrlowalowa, 2018-08-20:
Thank you for this information :) Now I have some more facts for my bachelor thesis.