siemens/meta-iot2000 #93: DHCP problem

deinok, 2018-07-25:
Hi, seems like DHCP is not working if a 3G mPCIe target is enabled, any idea of how to fix this issue?

jan-kiszka, 2018-08-14:
What exactly do you mean? Is DHCP not working for that 3G link? Or are you trying to use two network links, one over 3G, and the other (wired?) gets no IP via DHCP?

deinok, 2018-08-14:
I would like to use ETH1 with wire and DHCP enabled, and in case of network error fallback to PPP0 (3G).
But seems like if iot2000 detects 3g related hardware, DHCP stops working and no IPv4 is assigned to that interface

jan-kiszka, 2018-08-16:
Hmm, the iot2000setup tool should allow you to select multiple interface and also define dhcp for eth1 then. Did you do that?

deinok, 2018-08-16:
yes yes this was the first thing to try.
I also have tryed to start manualy `dhcpcd`.

deinok, 2018-08-16:
@jan-kiszka Yep, I also have configured that file and put the eth1 up with `ifup`.
I will investigate more, cause the strange thing is that this bug only happenz if Cellular is detected (Even if is not generating the ppp0). With a bit of luck, the support of SystemD (and indirectly NetworkD) can fix this issue.

mrlowalowa, 2018-08-19:
Just struggled with DHCP as well. Same hardware but new Image (with PCIe Wlan/Bluetooth driver installed but no PCIe hardware installed already)

Maybe it was my fritzbox or a similar issue you are referring to.
I solved it by locating the service in charge for DHCP on this image. The standard busybox dhcp program is udhcp.
With `$udhcp -i eth1 -r` I was able to get a new lease from my DHCP server (fritzbox). Restarting the DHCP Server and deleting its entry didn't help here.

Hope this helps you :small_airplane:

deinok, 2018-08-19:
@mrlowalowa Thanks, I will check it.

mrlowalowa, 2018-08-20:
but be careful with it. just issuing udhcpc or uhdcpd was killing the static address on eth0 until the next boot. I worked mainly over the serial debug interface for this.

Oh and I see that the command is wrong. You must use udhcpc (client) and I'm not 100% sure about the -r (new lease). It should not harm but if I remember it right I was just starting the client and it worked.

I try the same image now with an other DHCP server and will see if it was only a temporal solution or a permanent for this image. I sadly think it's just a temporal solution but therefore can be added in the autostart if you wish to have it consistent.

What is @jan-kiszka saying about this? Normally when I do a /etc/init.d/networking reboot I see the output: Sending DHCP request... etc but in this case I saw nothing. Is there any other way to confirm that a DHCP request was sent after reactivating a certain interface configured with dhcp?

jan-kiszka, 2018-08-20:
There isn't much black magic involved here, just the plain dhcp client from busybox and the settings in /etc/network/interfaces. Maybe there are servers that do not go well with that client, but I'm not aware of such issues. You may try to trace the traffic between client and server using wireshark (and a third machine attached to the same network, running its interface in promisc mode).

deinok, 2018-08-20:
@jan-kiszka @mrlowalowa I fixed the problem creating my own DHCP Client. Its a workaround, not a true fix.

Seems like the true fix should be move to another DHCP Client daemon.
http://git.yoctoproject.org/cgit.cgi/poky/tree/meta/recipes-connectivity/dhcp/dhcp_4.4.1.bb This should work.

If needed, I can generate a wireshark trace with the ICMP and DHCP traces

jan-kiszka, 2018-08-20:
What would be interesting for us (specifically before releasing the next version) is if this is A) a problem of busybox and B) if there is a fix we should apply or a busybox version we should update to. If A) is true but B) does not exist, moving to another client would be the alternative, yes.

deinok, 2018-08-20:
Is there a way that I can deactivate the busybox dhcpd so I can try to enable the dhcpd of poky without other issues?

mrlowalowa, 2018-08-21:
here is a way to manually choose your preferred dhcpc via the /etc/network/interfaces file @deinok :
https://unix.stackexchange.com/questions/261872/how-is-udhcpc-executed-and-how-to-change-it

And there seems to be a issue with udhcpc:
https://superuser.com/questions/1204231/busybox-udhcpc-can-not-be-killed
I'm experiencing as well a strange behavior of manually started udhcpc not be able to be killed. (but if I think more about it it might be because of the wrong execution of the program (all interfaces) which first broke my static IP address on eth0 and therfore also my ssh pipe on the 192.168.200.1 connection..)

maybe this next link sounds like a promising switch @jan-kiszka is asking for? -n for the configuration but it does not say the exact position where to place it in the config file..
https://unix.stackexchange.com/questions/267733/how-to-get-busybox-udhcpc-to-run-in-background-forever

I can reproduce my failure with udhcpc in the standard configuration. My FritzBox DHCP server is recognizing the MAC address of the IoT in the network but says it is passive and not using the connection. I'm unsure how I should use wireshark in easy way for this. ARP posining without an IP address of the target might not be the best solution and my switch is sadly not capable of packet mirroring (cheapest TP switch from amazon).

Probably should invest in some ninja stars.. https://hakshop.com/products/throwing-star-lan-tap
🤓

mrlowalowa, 2018-08-22:
This is working for me after rebooting:

```
auto eth1
iface eth1 inet dhcp
#       udhcpc_opts -S -R -n -p -r -i eth1
        pre-up /sbin/udhcpc -R -b -p /var/run/udhcpc.eth1.pid -i eth1
```
Just wanted to add that udhcpc_opts is also a possible way and has some effect (-S -> a bit more output to syslog) but sadly it just does not get me a IPv4 address. pre-up works like a charm..

Side question/fact/bug..?: Am I right that including vim to the image build gets rid of the classic vi ? I sadly noticed that I cannot select a section text and paste it with the middle mouse key i.e. into this browser field.
Problem solved.. Vim just includes/overwrites the link to vim.vim. So I have overwritten it now with a ln to the familiar busybox version of vi. Bug of vim or just a usage error.. I don't know.

jan-kiszka, 2018-08-24:
@mrlowalowa So, no changes to the pre-built images where needed, just fine-tuning of the interfaces file, right? We could add that to the pre-installed interfaces file, at least in commented-out form ("enable if you see dhcp problems").

Regarding vim / vi:
```
# ls -l /bin/vi
lrwxrwxrwx 1 root root 19 Aug 24 13:24 /bin/vi -> /bin/busybox.nosuid
```
It's the same story as with udhcpc. But if real vim does not overwrite anything of that mini vi, you should be fine.

mrlowalowa, 2018-08-26:
yes I just fine-tuned the interface and got an IP immediately. The only problem that still persists is that you do not get the other DHCP options without an additional ifup/ifdown <interface> on each startup. But this can be added to autostart and works fine this way.

thank you for the additional vi/vim information.

deinok, 2018-08-31:
@mrlowalowa Thanks for the hotfix. Works like a charm.

jan-kiszka, 2018-09-28:
I think we have a proper solution with that fix referenced above. Please test and let us know.

mauricek, 2018-10-31:
At least I can confirm that it fixed the DHCP issues on our setup. Thx for the fix.

closed 2018-10-31 by jan-kiszka