siemens/meta-iot2000 #94: Move to SystemD

deinok, 2018-08-13:
Due the fact that the SO of IOT seems to lack some common commands from Linux, I would like to suggest move to SystemD init system.

I know that SystemD have a bit of bad reputation for Linux purists, but this will really help IOT be more near to an expected Linux Board

jan-kiszka, 2018-08-14:
Switching to systemd is imaginable, but we haven't planned for that yet, specifically not for the next version. It would require rewriting some startup scripts and adjusting the iot2000setup tool. And then testing everything. You could support that effort by sending PRs, though.

deinok, 2018-08-14:
Just to ref: https://lists.yoctoproject.org/pipermail/yocto/2015-December/027882.html

The following local_conf should enable systemd:
```bb
DISTRO_FEATURES_append = " systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
```

Looks like there are some errors in compilation due init scripts.
Expecting .service and finding InitV scripts

deinok, 2018-08-14:
NVM, @jan-kiszka would you like to open a branch for systemd so I can send PR there?

Also, I think that I could create a `./meta-iot2000-examples/kas-systemd.yml` that extends the `./meta-iot2000-examples/kas.yml` but with systemd as init. While the transition is not fully tested.

jan-kiszka, 2018-08-16:
@deinok Just fork the project into your own namespace and issue PRs from there. That's a typical workflow pattern.

deinok, 2018-08-16:
Yes, i know the typical workflow, I'd just want to be sure that this kind of heavy changes are going to be accepted (If they work well :) ).

jan-kiszka, 2018-08-16:
If they work, I would be optimistic. The effort is likely as much in testing as it is in finding the places that need changes and applying them.

jan-kiszka, 2021-03-02:
Unlikely to happen for this layer anymore.

closed 2021-03-02 by jan-kiszka