siemens/meta-iot2000 #97: Is changing I2C bus speed possible?

HelloHungryImDad, 2018-08-25:
This is spawned from [this issue](https://github.com/siemens/meta-iot2000/issues/89) and my thread on the [Siemens forums](https://support.industry.siemens.com/tf/ww/en/posts/bmp280-sensor-and-iot2020/197037/?page=0&pageSize=10).

According to the [mraa documentation for Galileo gen 2](https://github.com/intel-iot-devkit/mraa/blob/master/docs/galileorevh.md) it is possible to change the I2C bus speed. However the example image here appears to use a different driver to the galileo board.

Is there any way to change the I2C bus speed on the IoT2020 using this image?

mrlowalowa, 2018-08-26:
Does the example image already feature the MRAA Python3 plugin?
I build a new image on my own and they linked the MRAA library now to the meta-openembedded layer like you can see here: https://github.com/siemens/meta-iot2000/issues/92

Maybe this new method already fixes yor problems? Have you just downloaded the example image or did you build your own image from the sources here? My idea to check for the Python3 mraa library is to check for this change to a newer source path. It's just my clue that this might indicate a newer version.

Do you have a tip for my how I could verify that the frequency has changed? I'm right now experiencing issues with my BNO055 sensor again. I use the Adafruit library and switched all smbus standard I2C libraries to the mraa part. But due to some electrical potential issues on my mobile robot the sensor crashes and needs a new start. I get an IOError in Python when my sensor crashes. continually trying to read again from the chip fails.
Surprisingly starting the python2.7 (old library) code again works. I'm not 100% sure about my situation. this situation happend with my build from earlier this year and with the old library. The same code that had the issue like you had with reading too much from the sensor.
Maybe I'm in this case not crashing my bus and only the sensor which is happy with a new connection (software I2C reset of the device).. I'll need to further investigate this.

I hope I can help you a bit with these information :)

nospam2000, 2018-08-26:
Your problem is probably not the bus speed. Please check out the following posting:
`https://forum.arduino.cc/index.php?topic=499612.0`:

> Another thing I found which is not clear in the datasheet - I2C "burst read" is not supported for the calibration registers - but works for the registers above 0xFC. Plus I found the genuine Adafruit module hung on burst read (would deliver the data, but then would not release the bus after the last byte, regardless of stop bits etc)

HelloHungryImDad, 2018-08-27:
Good find, thanks. That does sound like the root cause. Looks like I need to tweak the mraa code for this one then.

HelloHungryImDad, 2018-08-27:
After reviewing the mraa function for I2C communication, the [BMP280 datasheet](https://cdn-shop.adafruit.com/datasheets/BST-BMP280-DS001-11.pdf), and the [kernel documentation](https://www.kernel.org/doc/Documentation/i2c/i2c-protocol) for the i2c_dev driver I'm not sure that the master ack bit is the issue here.

The BMP280 sequence defined in the datasheet (page 30) is:
`Start Addr Write ACKS Data ACKS 
Start Addr Read ACKS Data ACKM Data NOACKM Stop`
and the i2c_dev driver (in read-write mode) does
`Start Addr Write ACKS Data ACKS
Start Addr Read ACKS Data ACKM Data NOACKM Stop`

So the sequence being sent by the master is apparently correct. Either the i2c_dev driver is not doing what it said it was doing or possibly still too high a clock?

jan-kiszka, 2018-08-28:
I don't remember we ever supported that bus speed tweaking for the legacy Intel BSP because it was not even available with the Yocto kernel that the first image version was still using. However, now we are on (almost) clean mainline, and there such hacks are gone.

The official way of changing the bus speed is tuning the ACPI table. That is, granted, not a trivial task. It would involve some fragment like https://github.com/westeri/meta-acpi/blob/master/recipes-bsp/acpi-tables/samples/joule/i2c1.asl, injected into our acpi fixups in meta-iot2000. I would have to find out the details as well.

Instead of diving into that, you can also try out via a hack if that speed is actually making the difference here: linux/drivers/mfd/intel_quark_i2c_gpio.c lists the two IOT2000 models and encodes the I2C default speed in driver_data. You could build your own kernel with that adjusted or feed a kernel patch via meta-iot2000 into a full image build.

HelloHungryImDad, 2018-08-29:
I'm certain at this stage that the clock speed is not the culprit. It is either the I2C driver or interface on the IoT2020 or the BMP280. 

At this stage it doesn't matter as I'll just write the burst read as a loop of incremented byte reads. Temperature doesn't change that quickly, and the extra overhead is minimal.

I'd be curious to find out what the issue is if anyone else encounters it.

Thanks for the responses.

closed 2018-08-29 by HelloHungryImDad