siemens/meta-iot2000 #98: Building an Image

Nick135, 2018-10-12:
Is there a better explanation how to build an Image?

The README.md don´t work for me.

Then I follow this example.
[https://www.dmcinfo.com/latest-thinking/blog/id/9395/creating-a-linux-image-for-the-siemens-iot2000-series](url)

yocto-2.1. ()
`
bitbake ERROR: Nothing PROVIDES iot2000-example-image
`
yocto-2.2.4 (morty):
` 
$ source poky/oe-init-build-env siemens-build

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'
~/siemens-build$ bitbake iot2000-example-image
Loading cache: 100% |############################################| Time: 0:00:01
Loaded 2250 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'mraa-utils' (but /home/osboxes/poky/meta-iot2000/meta-iot2000-example/recipes-core/images/iot2000-example-image.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'mraa-utils' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['mraa-utils']
ERROR: Required build target 'iot2000-example-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['iot2000-example-image', 'mraa-utils']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
`

I use a Virtual Box with Ubuntu_Mate_16.04.2-VB-64bit.

jan-kiszka, 2018-10-12:
Do not follow that cited web page for building meta-iot2000. It became outdated shortly after it was published, and unfortunately the author never reacted on my comment (see the bottom of this page).

Follow the instructions in our README. If you face issues there, let us know!

Nick135, 2018-10-12:
My VM has the packages requires for Yocto 2.2.
$ pip3 install kas

If I´m right, I only need this commands to build.
$ kas build meta-iot2000-bsp/kas.yml

How can I include mono in the build process?

jan-kiszka, 2018-10-12:
> If I´m right, I only need this commands to build.
> $ kas build meta-iot2000-bsp/kas.yml
> 

That will build you the basic bsp image, right. The full example image is defined by meta-iot2000-example/kas.yml.

> How can I include mono in the build process?

How to add a custom layer on top for adding own packages and modifying existing ones is discussed here: https://support.industry.siemens.com/tf/ww/en/posts/creating-custom-layer/182600/?page=0&pageSize=10. If you are lucky, mono is just about adding another 3rd-party yocto layer (http://layers.openembedded.org) and extending the package list in the image recipe. If you are unlucky, mono expects a processor with features that the Quark does not provide.

jan-kiszka, 2018-10-15:
> > If I´m right, I only need this commands to build.
> > $ kas build meta-iot2000-bsp/kas.yml
> 
> > > That will build you the basic bsp image, right. The full example image is defined by meta-iot2000-example/kas.yml.
> 
> Wenn ich das in der Konsole eingebe erhalte ich:
> `FileNotFoundError:` [Errno 2] No such file or directory: '/home/osboxes/meta-iot2000-example/kas.yml'
> Somit fehlt in der "README.md", dass man wie im Beispiel aus der Forum sich einen Ordner mit der kas.yml und ... anlegen muss.

[Please stay in English]

Would fb4fff531 help?

> 
> Anhand deines Beispieles hab ich ein Image erstellen können.
> Gibt es zu der kas.yml Datei eine Anleitung?

Yep, please see http://kas.readthedocs.io/

> Hab ein bisschen rum probiert und
> 
> > meta-mono:
> > url: git://git.yoctoproject.org/meta-mono
> > refspec: rocko

Usually, you want to specify a specific commit sha as refspec so that you can be sure to rebuild the same image the next time you try from scratch.

> 
> hinzugefügt. Sowie den Eintrag "IMAGE_INSTALL_append += "mono"" in der iot2000-example.image.bbappend.
> Somit hab ich es geschaft, dass der Eintrag für mono in die "bblayers.conf" gekommen ist und mono kompiliert wurde. War aber eher ein gebastel bis bitbake das gefressen hat.

Nick135, 2018-10-15:
> Would fb4fff5 help?
>
This helps to build the first image.

Normally I work behind a proxy. But I don't get it working.
I test with this in the kas.yml file:

```
proxy_config:
  http_proxy: http://User:Password@Address:Port
  https_proxy: https://User:Password@Address:Port
```

jan-kiszka, 2018-10-15:
We build behind proxies as well. It's ugly but it works in principle.

You will need to set these variables in the shell that invokes kas:
- http_proxy
- https_proxy
- ftp_proxy (there might be some cases remaining)
- no_proxy (can remain empty if there are no local repos)
- GIT_PROXY_COMMAND=gitproxy (gitproxy comes with yocto, needed for git:// downloads)

Setting the proxies in kas.yml is no longer recommended.

Nick135, 2018-10-22:
How can I add "apt-get" to the image?

I found this article, but when I run "kas build" the config will be overwritten.
[https://imxdev.gitlab.io/tutorial/How_to_apt-get_to_the_Yocto_Project_image/](url)

Nick135, 2018-10-22:
How can I add "apt-get" to the Image?

I found die article but the canges in the config will overriden by kas build
https://imxdev.gitlab.io/tutorial/How_to_apt-get_to_the_Yocto_Project_image/

Nick135, 2018-10-22:
aa

jan-kiszka, 2018-10-22:
Why do you want to add apt to your image? If that is supposed to enable the installation of packages from Debian or Ubuntu, then I need to tell you that this wont work. Or do you want to enable package-based update where the packages (in deb format) where built by Yocto before?

Nick135, 2018-10-22:
The plan was easy implementation of packages without bitbake.

> I need to tell you that this wont work
When it doesn't work, why is then the name of the article "How to apt-get to the Yocto Project image"?

Nick135, 2018-10-22:
Is there a USB driver in the "meta-iot2000-example"? Because I can't find the USB Stick plugged in the USB A Port of the IOT2040.

jan-kiszka, 2018-10-26:
Of course, you can add apt-get to your image an use that tool to install yocto-built packages in Debian-format onto the device. But you cannot install pre-built packages from standard Debian or Ubuntu this way.

Regarding USB: Check what `dmesg` reports when you plug the device. Also try to identify the device ID (Vendor:Device) via lsusb. That can help to find out if the device is supported at all by Linux, if we are just missing a configuration switch, or if you need a rather recent kernel version.

Nick135, 2018-10-30:
With `dmesg` I found the USB device. But it looks like the ID will be incremented with each plug/unplug.
How can I address the USB device like at windows 'd:/'

> [  783.821548] usb 1-1: USB disconnect, device number 2
> [  788.470174] usb 1-1: new high-speed USB device number 3 using ehci-pci
> [  788.638705] usb-storage 1-1:1.0: USB Mass Storage device detected
> [  788.651940] scsi host1: usb-storage 1-1:1.0
> [  789.656337] scsi 1:0:0:0: Direct-Access     Kingston DataTraveler 3.0      PQ: 0 ANSI: 6
> [  789.693913] sd 1:0:0:0: [sda] 30218842 512-byte logical blocks: (15.5 GB/14.4 GiB)
> [  789.728506] sd 1:0:0:0: [sda] Write Protect is off
> [  789.742685] sd 1:0:0:0: [sda] Mode Sense: 4f 00 00 00
> [  789.744130] sd 1:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
> [  789.797395]  sda: sda1
> [  789.830274] sd 1:0:0:0: [sda] Attached SCSI removable disk
> root@iot2000:~# lsusb
> Bus 001 Device 003: ID 0951:1666
> Bus 001 Device 001: ID 1d6b:0002
> Bus 002 Device 001: ID 1d6b:0001

jan-kiszka, 2018-10-31:
`mount /dev/sda1 /mnt` and, when done with it, `umount /mnt`.

Nick135, 2019-02-06:
How can I add p7zip? (meta-openembedded/meta-oe/p7zip)

jan-kiszka, 2019-02-07:
We already include meta-oe, so it is just about adding this package to the IMAGE_INSTALL variable. Can happen as local patch in meta-iot2000-example/recipes-core/images/iot2000-example-image.inc, or you add an overlay (custom) layer that extends it, see above.

Nick135, 2019-02-13:
I changed to an newer Linux and now I got this error:

> $ pip3 install kas
> Collecting kas
> Collecting jsonschema>=2.5.0 (from kas)
>   Using cached https://files.pythonhosted.org/packages/77/de/47e35a97b2b05c2fadbec67d44cfcdcd09b8086951b331d82de90d2912da/jsonschema-2.6.0-py2.py3-none-any.whl
> Collecting distro>=1.0.0 (from kas)
>   Using cached https://files.pythonhosted.org/packages/ea/35/82f79b92fa4d937146c660a6482cee4f3dfa1f97ff3d2a6f3ecba33e712e/distro-1.4.0-py2.py3-none-any.whl
> Collecting PyYAML>=3.0 (from kas)
> Installing collected packages: jsonschema, distro, PyYAML, kas
> Successfully installed PyYAML-3.13 distro-1.4.0 jsonschema-2.6.0 kas-0.19.0
> osboxes@osboxes:~/JE/meta-my-iot2000$ kas build meta-iot2000/meta-iot2000-example/kas.yml
> 
> Command 'kas' not found, did you mean:
> 
>   command 'kak' from deb kakoune
>   command 'cas' from deb amule-utils
>   command 'kar' from deb sra-toolkit
>   command 'kjs' from deb kdelibs-bin
>   command 'kash' from deb kanif
>   command 'kds' from deb kylin-display-switch
>   command 'as' from deb binutils
> 
> Try: sudo apt install <deb name>
> 
> osboxes@osboxes:~/JE/meta-my-iot2000$

jan-kiszka, 2019-02-14:
"pip3 install kas" - if run as normal user - seems to put the startup script for kas into a folder that is not in the normal search PATH. Simple workaround: sudo pip3 install kas.

Seems we need to tune our README...

Nick135, 2019-02-14:
The README worked for my Ubuntu Mate 16.4
With the Ubuntu Mate 18.4 "sudo -H pip3 install kas" works

jan-kiszka, 2019-02-15:
OK, great. We will update the README to address this pitfall as well.

closed 2019-02-15 by jan-kiszka