siemens/meta-iot2000 #101: Ubuntu Core image?

andrewebdev, 2018-10-16:
I would like to use Ubuntu Core on my device, and I don't want to use the image provided by siemens.
We need the the support, security and all the features provided by ubuntu snappy core.

It was recommended on this thread that I try to use the `intel nuc` as a starting point since that's also x86.
https://askubuntu.com/questions/1082399/ubuntu-core-image-for-iot2020

I installed the image but during startup in the `screen` session I just get to what looks like the firmware prompt. I suspect it doesn't recognize the image?

Is it possible to provide me with the basic details to create my own custom image for Ubuntu Snappy Core? (based on these instructions: https://docs.ubuntu.com/core/en/guides/build-device/image-building )

jan-kiszka, 2018-10-17:
Ubuntu depends on Debian, and Debian will never support the processor of the IOT2020/2040, see also https://support.industry.siemens.com/tf/ww/en/posts/debian-for-iot2000/162525/?page=0&pageSize=10.

If you are looking for robust firmware update technology: There will soon (within the next days) be an update of #75.

If you are interested in container technology, see https://support.industry.siemens.com/tf/ww/en/posts/experimental-docker-support/171337/?page=0&pageSize=10 (https://github.com/siemens/meta-iot2000/tree/jan/docker) and also https://support.industry.siemens.com/tf/ww/en/posts/mass-deployment-and-docker-support-resin-io-beta/195989/?page=0&pageSize=10.

What is unfortunately not available with free Yocto are long-term stable updates (there are commercial offerings). Will change with the next version of this device.

Nick135, 2018-10-22:
For tests I run my IOT2040 with this Debian Image.

[https://sourceforge.net/projects/iot2000-debian/](url)

Nick135, 2018-10-22:
For tests I use this Debian Image with my IOT2040.

[https://sourceforge.net/projects/iot2000-debian/](url)

Nick135, 2018-10-22:
For tests I run my IOT2040 with this Debian Image.

https://sourceforge.net/projects/iot2000-debian/

closed 2019-02-07 by jan-kiszka