siemens/meta-iot2000 #104: Node 8.x and Npm 6.x support

meriturva, 2018-10-30:
Hi to all,

i'm trying to build a new image with latest node 8.x version and npm 6.x version.

Actually i'm able to get node 8.9.4 working correctly, main problem is npm. In that case npm just return for every command an "Illegal instruction". I'm lost on that message and i really don't know how to solve it.

Are you going to support node 8.x instead of 6.x? I'm quite new to yocto builds and for me it is really time spending to build and test on physical device that everything works. Do you have any advice how to increase test phase? What do you use during yocto development to test image on correct architecture?

Thanks!

nospam2000, 2018-10-31:
The problem is, that there is no support for MMX instructions in the Quark CPU and the Google V8 JavaScript Engine which is used in Node.js no longer supports the x87 coprocessor.
The newest Node.js I could get working was the latest 7.x version.
I can provide more details if you need it.

meriturva, 2018-10-31:
So thanks @nospam2000 for you help!

I have made a short research and i found:

http://thread.gmane.org/gmane.comp.lang.javascript.v8.general/7797
https://chromium-review.googlesource.com/c/v8/v8/+/575756

I see that first releases of node 8.x could works because x87 support still on it. For example version 8.4.x... i tried and i get same "Illegal Function" error.

Am i right? are Node 8.4.x versions supporting x87 processor architecture (no MMX)?

nospam2000, 2018-10-31:
I tried some Node.js 8.x versions and even the oldest one didn't compile with the X87 switch, although officially the X87 support was dropped later.

jan-kiszka, 2018-11-21:
@meriturva Could you briefly summarize what all you tried? Just to avoid that someone else futilely retries that in the future. TIA!

meriturva, 2018-11-21:
Actually we have just discussed internally about x87 support on 8.x version and mainly we have tried (without success) to build latest version of 8.4.x just to be sure to have x87 support. No way, we had few problems and error during compile time ... but mainly we are not so happy to have a node 8.x version not updated. We think that for now it is important to have a secure LTS version working (and also maintained if necessary until April 2019). Going with node 8.x means no more update on 8.4 branch for sure!

For future we have to move to a different product and architecture due to CPU architecture limit (also dotnet is not supported on x87).

jan-kiszka, 2019-02-22:
See #107: We are almost there!