siemens/meta-iot2000 #105: nameserver problem caused by connman

tennfy, 2018-11-01:
Hi,

I built the latest image #1b74e27 yesterday. But I can't ping a domain after I connected the eth1 to internet.

I checked the /etc/resolv.conf  and found it is generated by connection manager. 

the content of /etc/resolv.conf is 

`# Generated by Connection Manager`
`nameserver ::1`
`nameserver 127.0.0.1`

It won't change with dhcp configuration, that's why I can't ping a domian.

Do you have any ideas to solve this problem?

nospam2000, 2018-11-01:
A general Linux rule is: files in /etc should not be automatically generated at runtime, such automatic generated files should go to /var/run and in /etc is softlink to them (see [Filesystem Hierarchy Standard](https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard)).
Can you look if you find an automatically generated file under /var/run or /var/lib?

jan-kiszka, 2018-11-01:
@tennfy But you did get an IP, didn't you?

@nospam2000 First of all, resolv.conf is different, maybe due to historic reasons. And second, moving things elsewhere won't solve the issue of connman fighting with the dhcp client.

nospam2000, 2018-11-01:
from the connman manpage:
```
--nodnsproxy
    Do not act as a DNS proxy. By default ConnMan will direct all DNS traffic to itself by setting
 nameserver to 127.0.0.1 in resolv.conf(5) file. If this is not desired and you want that all programs call
 directly some DNS server, then you can use the --nodnsproxy option. If this option is used, then
 ConnMan is not able to cache the DNS queries because the DNS traffic is not going through
 ConnMan and that can cause some extra network traffic. 
```

That means the content of /etc/resolv.conf is static and not adapted by DHCP when using connman in a default configuration. So the question is if connman is running and its commandline switches.

tennfy, 2018-11-01:
@jan-kiszka: yes, I can get an IP, DHCP client works.

After adding `--nodnsproxy` to /etc/init.d/connman and reboot, the content of /etc/resolv.conf changes to `nameserver 192.168.1.1`, It works well.

@nospam2000:  `connmanctl services` outputs nothing

tennfy, 2018-11-01:
I tried a 4G module in iot2000, but ofono can't detect it.

I have to use ppp to manually manage the 4G module, and it works.

It seems ofono is relatively old.

jan-kiszka, 2018-11-01:
We are on 1.20, poky master uses 1.24. Can you try pulling that version and check if it helps? I suspect copying over the ofono subtree from poky/meta/recipes/connectivity to meta-iot2000-example/recipes-connectivity will do that trick. Then we would consider a targeted update of that package.

For the DNS topic, I just pushed 6c8cefb4f07e0fe2e. Still needs to be checked against a working cellular setup.

tennfy, 2018-11-01:
@jan-kiszka How can I update the ofono version? I see ofono is configured in the `meta-iot2000-example/recipes-core/images/iot2000-example-image.inc`, there is no option to configure the version.

Will copy poky/meta/recipes/connectivity to meta-iot2000-example/recipes-connectivity work?

jan-kiszka, 2018-11-01:
That would be my expectation. But I just see that latest ofono is already 1.26 - as we are at it, give me minute...

mauricek, 2018-11-01:
Just stumbling over the same problem...

@nospam2000 Indeed, there is a symlink created from /etc/resolv.conf to /var/run/resolv.conf. However, after applying above "--nodnsproxy" the generated file only contains a comment, no other DNS set.

tennfy, 2018-11-02:
@jan-kiszka  I tried `f848039`，the version of ofono has updated to 1.26. but it still can't detect the 4G module.

jan-kiszka, 2018-11-02:
@tennfy OK, good to know. Then you may want to report this to upstream ofono.

closed 2018-11-02 by tennfy