siemens/meta-iot2000 #109: "'OPENSSL_1.0.0' not found" when using git

achih, 2018-11-22:
I want to run git on iot2040. But I got 'git: /usr/lib/libcrypto.so.1.0.0: version `OPENSSL_1.0.0' not found (required by git)' error message when I using git.

The git installed thought opkg install and repostory are following: 
src iotdk-all http://iotdk.intel.com/repos/2.0/iotdk/all
src iotdk-i586 http://iotdk.intel.com/repos/2.0/iotdk/i586
src iotdk-quark http://iotdk.intel.com/repos/2.0/iotdk/quark
src iotdk-x86 http://iotdk.intel.com/repos/2.0/iotdk/x86

Tested image: 
example-v2.2.0.wic -> from Siemens website.
build my own image from clone branch v2.2.x (shoud v2.2.1?)

Question:
Could you give me a little hint to make git work?

bergmanu, 2018-11-22:
Hi,
please do not use the Intel opkg repos. 
You can make git running by installing it from source. 

For git 2.17.0 for example use these commands: 

- curl -L https://github.com/git/git/archive/v2.17.0.tar.gz --output git-2.17.0.tar.gz
- tar xf git-2.17.0.tar.gz
- rm git-2.17.0.tar.gz
- cd git-2.17.0
- ./configure 
- make
- make install

achih, 2018-11-22:
Hi, bergmanu,
I'm trying your recommend.
many many thanks for your help. ^^

achih, 2018-11-22:
It perfectly works now~ ^^ 
Thank you so much.

closed 2018-11-22 by achih