siemens/meta-iot2000 #111: Creating Example IOT2000 image fails

karthikmuthineni, 2018-12-04:
I am building iot2000 example on Ubuntu 16.04. I have installed all the required packages for building image on host and met with all the requirements (50GB free space hard-disk). But, still I get errors during the process.

ERROR: Function failed: do_compile (log file is located at /home/karthik/build/tmp/work/i586-nlp-32-poky-linux/cmake/3.8.2-r0/temp/log.do_compile.10484)
ERROR: Task (/home/karthik/poky/meta/recipes-devtools/cmake/cmake_3.8.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4364 tasks of which 4362 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /home/karthik/meta-openembedded/meta-oe/recipes-extended/upm/upm_git.bb:do_compile
  /home/karthik/poky/meta/recipes-devtools/cmake/cmake_3.8.2.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

I had attached the log file. How to resolve this problem and  create image?

[log.do_compile.10484.txt](https://github.com/siemens/meta-iot2000/files/2642446/log.do_compile.10484.txt)

jan-kiszka, 2018-12-04:
Does it work when you retry building (without prior cleaning)?
Might also be a mismatch to the host build environment. Already considered using the kas docker container instead of native Ubuntu?

jan-kiszka, 2019-02-15:
Was the issue solved? Then we could close this.

karthikmuthineni, 2019-02-15:
Yes it's solved.

Thank you

closed 2019-02-15 by jan-kiszka