siemens/meta-iot2000 #114: Kas Build...failed with exit code '1'

mrsalhaoui, 2019-02-08:
I was trying to build th eimage using KAS 3 times, but still thiw error of NodeJS : 

| node_crypto_bio.cc:(.text+0x7ea): undefined reference to `BIO_clear_flags'
| collect2: error: ld returned 1 exit status
| make[1]: *** [/home/user/build/tmp/work/i686-linux/nodejs-native/6.14.3-r1.8/node-v6.14.3/out/Release/cctest] Error 1
| make[1]: Leaving directory `/home/user/build/tmp/work/i686-linux/nodejs-native/6.14.3-r1.8/node-v6.14.3/out'
| ERROR: oe_runmake failed
| make: *** [node] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/user/build/tmp/work/i686-linux/nodejs-native/6.14.3-r1.8/temp/log.do_compile.9275)
ERROR: Task (virtual:native:/home/user/meta-iot2000/meta-iot2000-example/recipes-devtools/nodejs/nodejs_6.14.3.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4279 tasks of which 4278 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/user/meta-iot2000/meta-iot2000-example/recipes-devtools/nodejs/nodejs_6.14.3.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

Im using ubuntu 14.04 in a virtual Machine.

Any Help please ?

jan-kiszka, 2019-02-09:
14.04 is likely just to old to build the initial steps, native packages. Switch to more recent release or - even safer- use the kas docker container.

mrsalhaoui, 2019-02-09:
Thank you for your request, which Ubuntu version you suggest ?

jan-kiszka, 2019-02-09:
No idea, I've no experience with Ubuntu for this setup. We are using Debian 9 (also in that container), and I also build on SUSE Leap 15.0.

mrsalhaoui, 2019-02-11:
I have tried to use another Virtual Machine using a new version of Ubuntu ( I tried 18.04.1), I found one in this website :  https://www.osboxes.org/ubuntu/
 I think is finished correctly this time 
[image]

jan-kiszka, 2019-02-11:
Looks good, indeed. So I'm closing the issue.

closed 2019-02-11 by jan-kiszka