siemens/meta-iot2000 #115: source poky building .... command not found

mrsalhaoui, 2019-02-12:
Hello,

I finished the Kas method with adding layers an packages, but When I try to make the source command it fails, any Help please ? 

[image]

jan-kiszka, 2019-02-12:
The kas build included that step already -  you are done with building! Just follow the README regarding  image writing.

mrsalhaoui, 2019-02-12:
but I have added  some layers amd packages after the Kas command : 
This is explaned by Bergmanu in this post : 

https://support.industry.siemens.com/tf/ww/en/posts/modifying-the-sd-card-image-adding-new-layers/177554/?page=0&pageSize=10

jan-kiszka, 2019-02-12:
Please follow the more recent description in https://support.industry.siemens.com/tf/ww/en/posts/creating-custom-layer/182600/?page=0&pageSize=10

mrsalhaoui, 2019-02-13:
It is not very clear, I will try too understand how it works again...!
But the changes that I have made after the kas command I have to delete them or not ?

jan-kiszka, 2019-02-13:
Yeah, that thread and the reference layer may lack some words if you aren't familiar with yocto yet.

The idea of the layer is to move you customizations in form of overlays and additional recipes into a separate repository. That allows to update meta-iot2000 without rewriting your changes or maintaining a fork of that layer. That's a typical yocto pattern.

Of course, you can also create branch in your meta-iot2000 clone, adjust what you would like to do differently or in addition, check that in and build the whole image as if you are building it unmodified from scratch. But then, when you want to use a newer version of meta-iot200 in the future, you will have to rebase your changes on top.

mrsalhaoui, 2019-02-13:
Here is my situation now : 
 I made the 3 next steps in the home folder : 
1.  install the packages required by yocto here https://www.yoctoproject.org/docs/2.2/mega-manual/mega-manual.html#packages 
2. pip3 install kas
3.  git clone https://github.com/siemens/meta-iot2000

then I have the home folder as shown  in the image : 

[image]

If I want to add this recipe  https://layers.openembedded.org/layerindex/recipe/47350/ which is in the openembedded-core Layer,  what should I do ? I have to clone a new layer in the home folder ? or just modify the tiny customization layer that jan-k proposed in this link 

https://support.industry.siemens.com/tf/ww/en/posts/creating-custom-layer/182600/?page=0&pageSize=10 ? 

Could you please describe to me clearly the changes that I have to do in order add this recipe before passing to the next steps ( source builb... and bitbake)

jan-kiszka, 2019-02-14:
The fastest solution to get to an image with only this additional package is indeed

- add ```IMAGE_INSTALL_append = " ffmpeg"``` to meta-iot2000-example/recipes-core/images/iot2000-example-image.inc
- run ```kas build meta-iot2000/meta-iot2000-example/kas.yml``` once again

Maintaining this little customization by re-adding this line to each newer version of our layer before bulding is fine.

mrsalhaoui, 2019-02-14:
After Making these Steps now, it appears another errors : 

* install the packages required by yocto here https://www.yoctoproject.org/docs/2.2/mega-manual/mega-manual.html#packages
* pip3 install kas
* git clone https://github.com/siemens/meta-iot2000
* kas build meta-iot2000/meta-iot2000-example/kas.yml
* add `IMAGE_INSTALL_append = " ffmpeg"` to meta-iot2000/meta-iot2000-example/recipes-core/images/iot2000-example-image.inc
* run `kas build meta-iot2000/meta-iot2000-example/kas.yml` once again

[image]

[image]

jan-kiszka, 2019-02-15:
OK, there is another trick needed: add ```LICENSE_FLAGS_WHITELIST += " commercial"``` to, e.g., meta-iot2000/meta-iot2000-example/conf/distro/poky-iot2000.conf.

mrsalhaoui, 2019-02-15:
Thank you Jan, 

The steps done until now  in order to enable ffmpeg are : 

* install the packages required by yocto here https://www.yoctoproject.org/docs/2.2/mega-manual/mega-manual.html#packages
* pip3 install kas
* git clone https://github.com/siemens/meta-iot2000
* kas build meta-iot2000/meta-iot2000-example/kas.yml
* add `IMAGE_INSTALL_append = " ffmpeg"` to meta-iot2000/meta-iot2000-example/recipes-core/images/iot2000-example-image.inc
add `LICENSE_FLAGS_WHITELIST += "commercial"` to, e.g., meta-iot2000/meta-iot2000-example/conf/distro/poky-iot2000.conf
* run `kas build meta-iot2000/meta-iot2000-example/kas.yml` once again

The result  after making the last command twice is : 

[image]

so now I don't need to use the source command and bitbake right ?  I have just to copy the image to the SD card using this command and test if the package ffmpeg works in the IoT gateway ?

* sudo dd if=build/tmp/deploy/images/iot2000/iot2000-example-image-iot2000.wic \
          of=/dev/mmcblk0 bs=4M oflag=sync

jan-kiszka, 2019-02-15:
Yes, it's exactly that easy now!

mrsalhaoui, 2019-02-15:
I want now to boot from an USD SD card reader, 
[image]
 but don't find the folder loader/entries, where can I find it ? what should I do to modify it to accept the boat from sdb/sdb1 as shown in the Picture.

mrsalhaoui, 2019-02-15:
Is It fine like that in Ubuntu ? 

[image]

jan-kiszka, 2019-02-15:
Not quite: Assuming that /dev/sdb is the SD card in your reader (you should check by validating that there is no /dev/sdb* when there is no reader), you should dump to the whole disk, ie. to /dev/sdb directly.

mrsalhaoui, 2019-02-15:
I have just zipped the .wic file and I writed the image on the SD card using windows, I can access now to the IoT gateway using SSH, but still the FFMPEG not working, I don't really know why ? is there another way to make it working, I really need it ..

jan-kiszka, 2019-02-15:
It may get way more complicated now: We are running on CPU here that is architecturally a Pentium I and has no SSE* accelerator instructions. Many modern x86 software use these instructions intensively, either by asking the compiler to inject them where reasonable - this is what we avoid when building applications via Yocto for the IOT2000 - or by using own assembly directly. The latter is not that easy to prevent. Sometimes it takes special configuration of the application build steps to activate a fallback, but often these fallbacks are not implemented any more. And sometimes they exist but became broken.

You could try running ffmpeg inside gdb on the target, to check if some illegal instruction or other exception occurs.

mrsalhaoui, 2019-02-15:
I have really no idea how to make that, any suggestions, please...

jan-kiszka, 2019-02-18:
So, some nodejs script is triggering the invocation of ffmpeg? Or what are your steps to trigger that error message?

mrsalhaoui, 2019-02-18:
I'm using a drone library in Github, the error came from the PngEncoder.js file, please see the link below : 

https://github.com/felixge/node-ar-drone/blob/master/lib/video/PngEncoder.js

jan-kiszka, 2019-02-18:
OK... So would have to extract the command line of the ffmpeg call (all parameters) from the that script, call ```gdb --args ffmpeg ..```, then tell gdb to ```run``` the program. When it crashed, issue ```bt``` and ```disassemble $pc,+10``` inside gdb. That may give us some idea what goes wrong. Not yet how to solve it, though.

mrsalhaoui, 2019-02-19:
1- Do You mean that I have to extract the command line of that error from the PngEncoder.js File ? or all the command lines of errors ? 

2- I have called the " gdb --args ffmpeg .. "" inside the video Folder that contains the PngEncoder.js File and the errors ? is that right, ? because I use other file in another folder to start the application and commincate with the drone ? 

I deleted the line of that error and I called the gdb inside Video folder, I have got these reults : 

[image]

jan-kiszka, 2019-02-20:
You need to find out, which options PngEncoder.js passes to ffmpeg when the problem occurs. That you need to feed into the gdb command I sketched.

mrsalhaoui, 2019-03-06:
Hello Again Jan,

I have contacted the person who made the code, and we tried to know what happend with FFMPEG : 

after changing a little bit the code, we found that type of the error : 

the error was SIGILL,  

"" means that FFMPEG has probably been compiled incorrectly. I.e. the binary contains a CPU instruction that isn't available on the CPU you're trying to execute this on.""

What do you think about that Jan ?

jan-kiszka, 2019-03-06:
This give a reasonable explanation for the crash, not yet for how the instruction(s) got into the binary. There are at least two options
- the recipe to build ffmpeg injects own switches that allow unsupported instructions
- the code contains inline assembly with such instructions

To get a clue about that, it would be good to have a backtrace to the instructions causing the SIGILL. Maybe your contact can help you with operating gdb in the required way - and also with interpreting the result. If it is a recipe issue, that can likely be fixed. Inline assembly would be tricky.

mrsalhaoui, 2019-03-06:
My contact said that I can debug/reproduce this by copying some h264 encoded video to this computer and trying to use FFMPEG to convert it into something else, 

I have tried to do that in the IOT gatway by using this command "ffmpeg -i video.flv image%d.jpg" to convert a video to images and it is working as you can see in the image,
[image]

About the debugger I don't really know how to use it yet,

jan-kiszka, 2019-03-07:
You still need to find out how exactly ffmpeg is called in your scenario. Depending on the selection mode, different code paths will be taken, and the case above may not trigger the corresponding ones. Check your original js file again. There is PngEncoder.prototype._spawnFfmpeg which assembles the command line of ffmpeg. Extract that, e.g. by printing ffmpegOptions, and try to reproduce the crash. Then we can discuss gdb usage again.

mrsalhaoui, 2019-03-13:
I tried as you proposed to print the ffmpegOptions in the PngEncoder.prototype._spawnFfmpeg function  : as you can see in the code below : 

PngEncoder.prototype._spawnFfmpeg = function() {
  var ffmpegOptions = [];
  ffmpegOptions.push('-i', '-'); 

  console.log("extraction-1",ffmpegOptions); // input flag

  ffmpegOptions.push('-f', 'image2pipe');

   console.log("extraction-2",ffmpegOptions);// format flag

  if(this._imageSize){
    ffmpegOptions.push('-s', this._imageSize);
    console.log("extraction-3",ffmpegOptions); // size flag
  }
  ffmpegOptions.push('-vcodec', 'png'); 

  console.log("extraction-4",ffmpegOptions);// codec flag

  ffmpegOptions.push('-r', this._frameRate);

  console.log("extraction-5",ffmpegOptions); // framerate flag

  ffmpegOptions.push('-');
   console.log("extraction-6",ffmpegOptions) // output

  // @TODO allow people to configure the ffmpeg path

  return this._spawn('ffmpeg', ffmpegOptions);

  console.log("extraction-7",ffmpegOptions);

};
 then I got these outputs after reproducing the same crash : 

[image]

jan-kiszka, 2019-03-13:
Do you have a .png file at hand? I think this should reproduce the crash as well:
```
ffmpeg -i some.png -f image2pipe -vcodec png -r 5 output.file
```
If yes, run that under gdb:
```
gdb --args ffmpeg ...
```

mrsalhaoui, 2019-03-14:
I think it works fine, no crash detected 

[image]

jan-kiszka, 2019-03-14:
If you look that the last line, you should see something familiar...

mrsalhaoui, 2019-03-14:
Illegal instruction ? What is that ?

jan-kiszka, 2019-03-14:
The same type of error message you js file was triggering as well.
So, now attach gdb to that ffmpeg call.

mrsalhaoui, 2019-03-14:
HEre we can see it also this after runing gdb 

[image]

jan-kiszka, 2019-03-14:
> OK... So would have to extract the command line of the ffmpeg call (all parameters) from the that script, call `gdb --args ffmpeg ..`, then tell gdb to `run` the program. When it crashed, issue `bt` [...]

This is what I was refererring to.

mrsalhaoui, 2019-03-14:
After an issue of bt and disassemble $pc,+10 I have got : 

[image]

mrsalhaoui, 2019-03-14:
What do you think Jan ?

jan-kiszka, 2019-03-14:
cmov is an unsupported instruction for the target CPU. I'm trying to rebuild the recipe to check why it was used by ffmpeg. But my environment is already some versions ahead, may not give identical results.

Could you attach (no screenshots, please) the log file from the build similar to ```build/tmp/work/i586-nlp-32-poky-linux/ffmpeg/3.3.3-r0/temp/log.do_configure```?

mrsalhaoui, 2019-03-14:
Ok please find attached the file

jan-kiszka, 2019-03-14:
Yeah, that's in line with what I see. Here is a quick-fix/hack:

Create ```meta-iot2000-bsp/recipes-multimedia/ffmpeg/ffmpeg_%.bbappend``` with this content:
```
def cpu(d):
    return "i586"
```

Then rebuild. The log.do_configure should then show that MMX, SSE, i686 features etc. are all disabled, and the crash should be gone.

mrsalhaoui, 2019-03-14:
There is no meta-iot2000-bsp, I have just meta-iot2000 in my machine ?

jan-kiszka, 2019-03-14:
```mkdir -p meta-iot2000/meta-iot2000-bsp/recipes-multimedia/ffmpeg```, then create that file.

mrsalhaoui, 2019-03-14:
ok the file is created, so do you mean I have to type the Kas command again right : 

kas build meta-iot2000/meta-iot2000-example/kas.yml

jan-kiszka, 2019-03-14:
Yes, that should trigger the rebuild of ffmpeg (you will see that) and the re-assembly of the image.

mrsalhaoui, 2019-03-14:
Ok Thank you Jan, I'm making it now, I will put this new image in the SD card and tell you if the application works fine.

mrsalhaoui, 2019-03-15:
All of this was to make ffmpeg working perfectly in Siemens IOT2040 for this application,

Thank you Jan, I can now see the pictures from the drone

[image]

jan-kiszka, 2019-03-15:
Hey, cool! Congrats!

I will add at least that ffmpeg recipe artifact to the bsp layer, for future releases.

closed 2019-03-15 by jan-kiszka