siemens/meta-iot2000 #118: Running iot2000 image inside docker

tamaramueller, 2019-03-18:
Hi,

I followed the instruction of the README file and successfully finished the Docker build of the bsp image. I am working on Ubuntu 18.04. Now I would like to run the built image inside the docker container to simulate an iot2000 device. Therefore I tried  `source oe-init-build-env` and afterwards `runqemu iot2000`. I get the following error messages:

```
runqemu - INFO - Running MACHINE=iot2000 bitbake -e...
runqemu - ERROR - /shared-volume/poky/build/tmp/deploy/images/iot2000 not a directory valid DEPLOY_DIR_IMAGE
ls: cannot access '/shared-volume/poky/build/tmp/deploy/images/iot2000/*.qemuboot.conf': No such file or directory
runqemu - ERROR - Command 'ls -t /shared-volume/poky/build/tmp/deploy/images/iot2000/*.qemuboot.conf' returned non-zero exit status 2
Cleanup
```

Is there a way to simulate the iot device inside the Docker container? Does the host operating system (Ubuntu 18.04) matter?

I am completely new to Yocto and Docker, so there might be just a beginner mistake somewhere.

Thanks in advance for suggestions!

jan-kiszka, 2019-03-18:
Were you building current master (ff8c85ff6f7a29686920b42491e7854f97893793)? The runqemu thing once worked, but I suppose it was never tested inside the kas docker image. And it might have been broken since 6dc079110bb8e9538c509b7d6eaab495ba2fd818.

tamaramueller, 2019-03-18:
Yes, I am using the current master. Should I try using an older commit?

jan-kiszka, 2019-03-18:
If you have the time (will trigger a rebuild...), you could try that initial commit for QEMU support.

We definitely do not have a qemuboot.conf yet. Will need one for a major uptdate, but it seems as if the need for that was also introduced in the middle of the yocto 2.4.x stable series.

jan-kiszka, 2019-03-18:
Wait, maybe it's just how to call runqemu: https://www.yoctoproject.org/docs/2.4/ref-manual/ref-manual.html#migration-2.2-runqemu-ported-to-python.

tamaramueller, 2019-03-19:
Thanks for your quick response!

I tried using an old commit and using a command like in the link you sent me, but unfortunately I couldn't get either to work.

I tried 
```
runqemu -d build/tmp/deploy/images/iot2000/core-image-minimal-iot2000.wic
```
and 
```
runqemu /shared-volume/build/tmp/deploy/images/iot2000 /shared-volume/build/tmp/deploy/images/iot2000/core-image-minimal-iot2000.ext4 /shared-volume/build/tmp/deploy/images/iot2000/bzImage nographic
```
and got the following error message:
```
runqemu - ERROR - TUN control device /dev/net/tun is unavailable; you may need to enable TUN (e.g. sudo modprobe tun)
```
I couldn't find a way to enable TUN, do have an idea how that works?

Thanks a lot!

xenomai-ci, 2019-03-19:
Did you try the suggested ```sudo modprobe tun```?

jan-kiszka, 2019-03-19:
Did you try the suggested ```sudo modprobe tun```?

tamaramueller, 2019-03-19:
Yes, that didn't work:
```
bash: modprobe: command not found
```

jan-kiszka, 2019-03-19:
Argh, we are inside a container, and there you have no access to that interface by default. Try adding ```slirp``` to the runqemu command, see also https://www.yoctoproject.org/docs/2.6.1/dev-manual/dev-manual.html#qemu-dev-runqemu-command-line-options

tamaramueller, 2019-03-19:
Awesome! Thanks a lot, that works

closed 2019-04-17 by jan-kiszka