siemens/meta-iot2000 #119: release 2.4.0 build error

ThinkontrolSY, 2019-03-21:
build in docker

```sh
2019-03-21 13:26:11 - ERROR    - [Errno 2] No such file or directory: '/shared-volume/meta-iot2000/meta-iot2000-example/meta-iot2000-bsp/kas.yml'
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/kas/kas.py", line 161, in main
    sys.exit(kas(sys.argv[1:]))
  File "/usr/local/lib/python3.5/dist-packages/kas/kas.py", line 149, in kas
    if plugin().run(args):
  File "/usr/local/lib/python3.5/dist-packages/kas/build.py", line 110, in run
    macro.run(ctx, args.skip)
  File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 63, in run
    command.execute(ctx)
  File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 106, in execute
    while all(executor(c) for c in self.commands):
  File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 106, in <genexpr>
    while all(executor(c) for c in self.commands):
  File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 104, in executor
    return command.execute(ctx)
  File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 312, in execute
    ctx.config.handler.get_config(repos=repo_paths)
  File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 259, in get_config
    cfgs, reps = _internal_include_handler(configfile)
  File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 208, in _internal_include_handler
    includefile)))
  File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 170, in _internal_include_handler
    current_config = load_config(filename)
  File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 62, in load_config
    with open(filename, 'rb') as fds:
FileNotFoundError: [Errno 2] No such file or directory: '/shared-volume/meta-iot2000/meta-iot2000-example/meta-iot2000-bsp/kas.yml'
```

jan-kiszka, 2019-03-21:
Please share the steps you did to trigger this.

ThinkontrolSY, 2019-03-22:
I just follow the step in readme
1. download from release v2.4.0 source code
2. unzip package
3. run docker
   ```sh
   docker run -v $(pwd):/shared-volume:rw -e USER_ID=$(id -u) -e https_proxy=$https_proxy -e http_proxy=$http_proxy -t -i --storage-opt size=50G kasproject/kas:latest
   ```
4. in docker
   ```sh
   builder@476304c867c5:/shared-volume$ kas build meta-iot2000-2.4.0/meta-iot2000-example/kas.yml
   ```
5. than I get above error
   ```sh
   2019-03-22 11:30:23 - INFO     - kas 1.0 started
   2019-03-22 11:30:23 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
   2019-03-22 11:30:23 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
   2019-03-22 11:30:23 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
   2019-03-22 11:30:23 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
   2019-03-22 11:30:23 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-iot2000
   2019-03-22 11:30:23 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
   2019-03-22 11:30:23 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
   2019-03-22 11:30:24 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-openembedded
   2019-03-22 11:30:24 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
   2019-03-22 11:30:24 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
   2019-03-22 11:30:24 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-iot2000
   2019-03-22 11:30:24 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
   2019-03-22 11:30:24 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
   2019-03-22 11:30:24 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-openembedded
   2019-03-22 11:30:24 - ERROR    - [Errno 2] No such file or directory: '/shared-volume/meta-iot2000-2.4.0/meta-iot2000-example/meta-iot2000-bsp/kas.yml'
   Traceback (most recent call last):
     File "/usr/local/lib/python3.5/dist-packages/kas/kas.py", line 161, in main
       sys.exit(kas(sys.argv[1:]))
     File "/usr/local/lib/python3.5/dist-packages/kas/kas.py", line 149, in kas
       if plugin().run(args):
     File "/usr/local/lib/python3.5/dist-packages/kas/build.py", line 110, in run
       macro.run(ctx, args.skip)
     File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 63, in run
       command.execute(ctx)
     File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 106, in execute
       while all(executor(c) for c in self.commands):
     File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 106, in <genexpr>
       while all(executor(c) for c in self.commands):
     File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 104, in executor
       return command.execute(ctx)
     File "/usr/local/lib/python3.5/dist-packages/kas/libcmds.py", line 312, in execute
       ctx.config.handler.get_config(repos=repo_paths)
     File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 259, in get_config
       cfgs, reps = _internal_include_handler(configfile)
     File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 208, in _internal_include_handler
       includefile)))
     File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 170, in _internal_include_handler
       current_config = load_config(filename)
     File "/usr/local/lib/python3.5/dist-packages/kas/includehandler.py", line 62, in load_config
       with open(filename, 'rb') as fds:
   FileNotFoundError: [Errno 2] No such file or directory: '/shared-volume/meta-iot2000-2.4.0/meta-iot2000-example/meta-iot2000-bsp/kas.yml'
   ```

jan-kiszka, 2019-03-22:
Try a cd into the unzipped meta-iot2000 first. I'm afraid that is an undocumented requirement (needs to be clarified and ideally also relaxed eventually).

ThinkontrolSY, 2019-03-23:
> Try a cd into the unzipped meta-iot2000 first. I'm afraid that is an undocumented requirement (needs to be clarified and ideally also relaxed eventually).

I cd into meta-iot2000-2.4.0 before build, but still get the same error.
than I checked the kas.yml file, It seems path is incorrect, the original content:
```yml
header:
  version: 2
  includes:
    - file: meta-iot2000-bsp/kas.yml
      repo: meta-iot2000
```
but `meta-iot2000-bsp` is outside the `meta-iot2000-example`, so I change the file:
```yml
header:
  version: 2
  includes:
    - file: ../meta-iot2000-bsp/kas.yml
      repo: meta-iot2000
```
the above error gone;

than another error shows:
```sh
2019-03-23 08:12:40 - INFO     - Repository meta-nodejs cloned
2019-03-23 08:13:06 - INFO     - Repository meta-intel cloned
2019-03-23 08:14:58 - INFO     - Repository poky cloned
2019-03-23 08:16:35 - INFO     - Repository meta-openembedded cloned
2019-03-23 08:16:35 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
2019-03-23 08:16:35 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
2019-03-23 08:16:35 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-iot2000
2019-03-23 08:16:35 - INFO     - /shared-volume/meta-openembedded$ git status -s
2019-03-23 08:16:35 - INFO     - /shared-volume/meta-openembedded$ git rev-parse --verify HEAD
2019-03-23 08:16:35 - INFO     - 5ea50a04748e7caa4974bdbe65ff5df6cb27fc8d
2019-03-23 08:16:35 - INFO     - /shared-volume/meta-openembedded$ git checkout -q eae996301d9c097bcbeb8046f08041dc82bb62f8
2019-03-23 08:16:36 - INFO     - /shared-volume/meta-intel$ git status -s
2019-03-23 08:16:36 - INFO     - /shared-volume/meta-intel$ git rev-parse --verify HEAD
2019-03-23 08:16:36 - INFO     - c7c0e12c98cec527bcdcf26594d34bbeb16057c5
2019-03-23 08:16:36 - INFO     - /shared-volume/meta-intel$ git checkout -q 4ee8ff5ebe0657bd376d7a79703a21ec070ee779
2019-03-23 08:16:36 - INFO     - /shared-volume/meta-nodejs$ git status -s
2019-03-23 08:16:36 - INFO     - /shared-volume/meta-nodejs$ git rev-parse --verify HEAD
2019-03-23 08:16:36 - INFO     - 0b32420ef0712f53165c3ad90eaf670af302d1f4
2019-03-23 08:16:36 - INFO     - /shared-volume/meta-nodejs$ git checkout -q 7e90d506488036cf1c490d1ac1086675eb2805b6
2019-03-23 08:16:36 - INFO     - /shared-volume/poky$ git status -s
2019-03-23 08:16:36 - INFO     - /shared-volume/poky$ git rev-parse --verify HEAD
2019-03-23 08:16:36 - INFO     - 2187d06166ed6819a1d07c3f8a3be7a256fab18c
2019-03-23 08:16:36 - INFO     - /shared-volume/poky$ git checkout -q 2731fd35d5570263bc004ab16acf4acb0d94422b
2019-03-23 08:16:37 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
2019-03-23 08:16:37 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
2019-03-23 08:16:37 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-iot2000
2019-03-23 08:16:37 - INFO     - /shared-volume/poky$ /tmp/tmpg0fx1iaw /shared-volume/build
2019-03-23 08:16:37 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
2019-03-23 08:16:37 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
2019-03-23 08:16:37 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-iot2000
2019-03-23 08:16:37 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ git rev-parse --show-toplevel
2019-03-23 08:16:37 - INFO     - /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example$ hg root
2019-03-23 08:16:37 - INFO     - Using /shared-volume/meta-iot2000-2.4.0/meta-iot2000-example as root for repository meta-iot2000
2019-03-23 08:16:37 - INFO     - /shared-volume/build$ /shared-volume/poky/bitbake/bin/bitbake -k -c build iot2000-example-image
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/shared-volume/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 202 at 2019-03-23 08:16:38.056105 ---
ERROR: Layer directory '/shared-volume/meta-iot2000-2.4.0/meta-iot2000-example/meta-iot2000-bsp' does not exist! Please check BBLAYERS in /shared-volume/build/conf/bblayers.conf

builder@0601dd38f87e:/shared-volume$ 
```

jan-kiszka, 2019-03-23:
Ah, now I recall: Building from a tarball is not supported right now (and was rarely requested so far anyway). That's because kas cannot identify the repo root in the absence of git as an anchor. We could probably enhance kas with some search algorithm if that's really needed.

ThinkontrolSY, 2019-03-23:
> e

OK, I see, Thanks
How to checkout 2.4.0? on which branch
the reason I need 2.4.0 is that **Add cellular modem support**, Can you list the supported cellular modem?

jan-kiszka, 2019-03-24:
If you clone the repo, you will be on master branch which is (almost) equivalent to 2.4.0.

You can roughly derive the list of supported devices from https://git.kernel.org/pub/scm/network/ofono/ofono.git/tree/doc/hardware-support.txt?h=1.26, but it can be tricky to identify them upfront, specifically as vendors tend to not reveal the used chipset in a specific model (may even change for an existing model during production time).

ThinkontrolSY, 2019-03-24:
> If you clone the repo, you will be on master branch which is (almost) equivalent to 2.4.0.
>
> You can roughly derive the list of supported devices from [...]

Very helpful

jan-kiszka, 2019-05-28:
master is now also supporting the kas build from tarballs - finally closing this.

closed 2019-05-28 by jan-kiszka