siemens/meta-iot2000 #121: Update node-red 0.20+

Themesaul, 2019-04-16:
Would it be possible to update the node-red version to a current version, for example version 0.20?

closed 2019-04-16 by Themesaul

jan-kiszka, 2019-04-16:
I suspect you found https://github.com/siemens/meta-iot2000/commits/jan/thud

Themesaul, 2019-04-17:
> I suspect you found https://github.com/siemens/meta-iot2000/commits/jan/thud

Yes hahahaha . jan/thud is incredible. I'm trying it now. Do you think it's stable? Why does not it belong to the master branch?

jan-kiszka, 2019-04-17:
It needs further testing and also reviews before it can go into master. And I only finished the last update yesterday. So, any early testing and feedback would be very welcome!

Themesaul, 2019-04-17:
> It needs further testing and also reviews before it can go into master. And I only finished the last update yesterday. So, any early testing and feedback would be very welcome!

After using it I see that it works very well. Node-red is noticeably faster than the master's version. Especially when you press the "deploy" button. I have added the redis, git, sysstat, etc layers and everything works fine. What kinds of tests should be done? What could go wrong in this branch? thx!

jan-kiszka, 2019-04-17:
That sounds great already!

Regarding node-red, the highest risk is sure my forward port of the quark patches to node.js 8.x. E.g. I had to do this: https://github.com/siemens/meta-iot2000/blob/jan/thud/meta-iot2000-bsp/recipes-devtools/nodejs/nodejs/0003-Relax-check-in-ComputeInputFrameSize.patch. So, running node-red, adding / removing nodes from the palette etc. is important.

In general, that branch is a major update. Most packages have been refreshed by puling a new yocto version. So, every corner can have smaller or larger sharp edges remaining.

Themesaul, 2019-04-18:
> That sounds great already!
>
> Regarding node-red, the highest risk is sure my forward port of the quark patches to node.js 8.x. E.g. I had to do this: [...]

I have installed nodes with npm, I have deployed flows, I have done more tests and everything works as it should. I'm going to leave the system for 6 days sending data, but it seems to work very well. Good job!