siemens/meta-iot2000 #122: modbus nodejs 8

Themesaul, 2019-05-08:
After trying the compilation of NodeJS 8 for several days everything has gone perfect. Now, I'm trying to use node-red-contrib-modbus. The node is added correctly, but when I try to use it and deploy it, it kills the node-red service. I have seen that this same problem of "Illegal instruction" is in the Siemens forum with the example image and that they released a new version where they fixed it, so I do not think it is the fault of NodeJs 8. Does anyone know what I can do ?

Log:

``` sh
Welcome to Node-RED
===================
...
...
8 May 07:15:09 - [info] Starting flows
8 May 07:15:36 - [info] Started flows
Illegal instruction
root@iot2000:~#
```

jan-kiszka, 2019-05-08:
Could you check if that node contains binary bindings? That would be a simple explanation for a crash. Then a npm rebuild --build-from-source may help (and we should pre-package that node to avoid that manual step - was already considering this). 

Otherwise, attaching gdb to the node instance running node-red would be good so that we get a backtrace when it crashes. 

BTW, I'm just building an update with nodejs 8.16.0.

Themesaul, 2019-05-08:
--build-from-source  work  like a charm! Thx

nodejs 8.16.0. <3 Good job!

Themesaul, 2019-05-08:
###

closed 2019-05-08 by Themesaul

jan-kiszka, 2019-05-08:
Yeah, I think I know why: this package depends on serialport, just like node-red-node-serialport, and it will likely pull its own version when installing. I will add it to our package set to avoid this.