siemens/meta-iot2000 #123: Add node-red-contrib-s7

tennfy, 2019-05-09:
node-red-contrib-s7 is also widely used, maybe we could include it in the image.

jan-kiszka, 2019-05-09:
We will discuss this. Two questions to support this:
- Is it specifically complicated (maybe due to binary parts) to install this node from Node-RED on a running IOT2000?
- What's your practical experience regarding the maturity / compatibility of this node?

tennfy, 2019-05-10:
1. As I know, it is pure js, no binary package is needed
2. I have tested on s7-200 smart and s7-1500, there is no problem. For s7-400, there is a bug with rack/slot mode and user has to use asap mode.

jan-kiszka, 2019-05-28:
For now, we will not include this node, specifically as it is very easy to install manually when needed.

closed 2019-05-28 by jan-kiszka