siemens/meta-iot2000 #129: Image creation fails: double free detected in tcache 2 (Ubuntu 18.04.2 / docker)

mvalkama, 2019-06-14:
Just got the device last week and been trying to get it working with a display that has NodeJS library available. Unfortunately it requires at least nodejs 7.6, so been trying to build a new image for it. 

After few attempts I finally got the docker build to almost pass. Just at the end when it seems to be creating the image files, I run into this "double free detected" error for both minimal and example image, logs attached. Native build didn't progress this far.

`ERROR: core-image-minimal-initramfs-1.0-r0 do_rootfs: Unable to install packages. Command '/shared-volume/build/tmp/work/iot2000-poky-linux/core-image-minimal-initramfs/1.0-r0/recipe-sysroot-native/usr/bin/opkg ...' returned -6: free(): double free detected in tcache 2`

Not that familiar with docker, but tweaked the settings a bit to get it working. Removed storage-opt argument and added /etc/docker/daemon.json to enable overlay2 driver as there was some complaint about that) 

Here's the command I run (same as in readme, except git removed after first run attempt and debug flag added)
docker run -v $(pwd):/shared-volume:rw -e USER_ID=$(id -u) --rm -t -i kasproject/kas:latest sh -c "cd /shared-volume && kas -d build meta-iot2000/kas-example.yml"

Should I try even older Ubuntu or what could cause this? 

[example_log.do_rootfs.txt](https://github.com/siemens/meta-iot2000/files/3290235/example_log.do_rootfs.txt)
[minimal_log.do_rootfs.txt](https://github.com/siemens/meta-iot2000/files/3290236/minimal_log.do_rootfs.txt)

jan-kiszka, 2019-06-14:
Haven't seen this error before. Which revision of meta-iot2000 did you build? What was the docker version?

You may also try a native build, ie. install the required packages for yocto builds under your ubuntu and the run kas without docker.

mvalkama, 2019-06-14:
>Which revision of meta-iot2000 did you build? 

master.

>What was the docker version?

`Docker version 18.09.5, build e8ff056`

>You may also try a native build, ie. install the required packages for yocto builds under your ubuntu and the run kas without docker.

Tried that already, didn't get this far with that. IIRC it failed on gcc compile. Tried both also on Ubuntu 19.04 but got some failures on that as well, so reinstalled 18.04.2 and with docker got all compiles to pass.

jan-kiszka, 2019-06-14:
Hmm, if you build master, you should have 'kas build kas-example.yml' in your command line. Something is inconsistent here.

jan-kiszka, 2019-06-14:
Alternative docker invocation:
```
wget https://raw.githubusercontent.com/siemens/kas/master/kas-docker
chmod a+x kas-docker
./kas-docker build kas-example.yml
```
We will update the documentation soon.

mvalkama, 2019-06-19:
No luck so far, been trying with kas-docker style and ran into another problem. With this the perl-native build fails due to some "read-only" filesystem. 

`Writing Makefile for base
Unable to open MakeMaker.tmp: Read-only file system at /repo/build/tmp/work/x86_64-linux/perl-native/5.24.4-r0/perl-5.24.4/cpan/ExtUtils-MakeMaker/lib/ExtUtils/MakeMaker.pm line 1173.`

[log.do_compile.txt](https://github.com/siemens/meta-iot2000/files/3304536/log.do_compile.txt)

jan-kiszka, 2019-06-19:
Just tried `./kas-docker build meta-iot2000/kas-example.yml --target perl-native` locally from scratch - no issues here. Will look for someone using Ubuntu or install an own VM. Docker was taken from Ubuntu or from the Docker repo?

mvalkama, 2019-06-19:
> Will look for someone using Ubuntu or install an own VM. 

Thanks.

> Docker was taken from Ubuntu or from the Docker repo?

Ubuntu. The installation is pretty much default installation of Ubuntu 18.04.2 with just the necessary packages installed that were listed on Yocto pages plus docker and kas.

jan-kiszka, 2019-06-19:
Everything working as expected: I did a minimal installation of Ubuntu 18.04.2 from the ISO image, then just added docker.io and added my user to the docker group, and then I was able to build perl-native as above.

mvalkama, 2019-06-19:
Hmm.. Have to check the docker group status once I get back to that machine this evening. I might have forgotten that from this attempt.

jan-kiszka, 2019-06-19:
If you are not in the docker group, kas-docker (namely the docker command it spawns) will not start at all.

I only created a clean VM image (KVM) for Ubuntu. Maybe yours had some other changes already that cause these issues.

mvalkama, 2019-06-21:
Got it working. Forgot "tiny" detail, I had been using Ubuntu Server instead of Desktop. Installed 18.04.2 desktop version and didn't try to run everything over ssh and build passed on first attempt. 
I did increase swap size considerably (machine has 8GB of RAM) before starting. On one attempt on server installation I noticed that default allocation was not enough. 
Thanks for your patience!

closed 2019-06-21 by mvalkama