siemens/meta-iot2000 #134: EFI partition not building correctly

davidhgriffiths, 2019-08-14:
Building the iot2000-example image using Docker proceeds without reporting any errors but the EFI partition appears to have built incorrectly.  It is titled Boot0 instead of EFI and contains only the following four files and no EFI or loader directory:
acpi-upgrades-iot2000.cpio
BGENV.DAT
bzimage
EFILABEL

The image will not boot on an IoT2020.

jan-kiszka, 2019-08-15:
I assume you were building latest master, right? This will generate a disk image with 2 FAT partitions. One of them is the 2nd-stage boot partition you looked at (partition 2, "boot0"), the other is the actual EFI partition (partition 1, "efi"). It contains EFI/BOOT/bootia32.efi, the bootloader. Should look like this:

```
$ /sbin/sfdisk -l build/tmp/deploy/images/iot2000/iot2000-example-image-iot2000.wic
Disk build/tmp/deploy/images/iot2000/iot2000-example-image-iot2000.wic: 2.3 GiB, 2409866240 bytes, 4706770 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 3AA79466-455A-46AD-9E3E-CAD66247C50D

Device                                                               Start     End Sectors  Size Type
build/tmp/deploy/images/iot2000/iot2000-example-image-iot2000.wic1    2048   67583   65536   32M EFI System
build/tmp/deploy/images/iot2000/iot2000-example-image-iot2000.wic2   67584  133119   65536   32M Microsoft basic data
build/tmp/deploy/images/iot2000/iot2000-example-image-iot2000.wic3  133120 1181695 1048576  512M Linux swap
build/tmp/deploy/images/iot2000/iot2000-example-image-iot2000.wic4 1181696 4706735 3525040  1.7G Linux filesystem
```

davidhgriffiths, 2019-08-15:
Thanks for the quick response.   I missed the EFI partition because it had not mounted automatically on my system.  However,  I still can't find the /loader/entries/boot.conf file or either of the folders in its path on any of the partitions.

jan-kiszka, 2019-08-16:
There is a different boot loader in use, EFi Boot Guard, that does not use such a file. This boot loader plays an important role when you build the swupdate image that allows safe remote updates, and to simplify the setup, we also installed it for the normal example image.

Do you need to modify the boot parameters?

glcj, 2020-08-25:
Hello,

How can I switch the boot to USB drive with EFI?

Thanks,

jan-kiszka, 2020-08-25:
By providing a bootable image via USB and not doing that via the SD card.

jan-kiszka, 2020-08-25:
Correction: This does not work, unfortunately. On the IOT2050, we already switched to booting scheme where the device name (/dev/mmcblkX, /dev/sdaY) does not matter, but the IOT2000 does not have this yet. Let me fix that...

glcj, 2020-08-25:
Hi Jan,

I will be attentive to perform the tests,

Try partitioning the file "wic-image.iot2000.wks" and the header file.

But the boot process always ends in:

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Tainted: G       A        4.4.208-cip41 #1
Hardware name: Intel Corp. QUARK/SIMATIC IOT2000, BIOS V24.01.01 01/01/2014
 f4c99f40 f4c99f40 f4c99ef8 c12a8182 f4c99f10 c10e2c54 fffffffa f4c99f40
 fffffffa c0080000 f4c99f6c c1847e48 c16df91c f4c99f40 c16df8c8 c16df894
 c18880c5 f4c99f40 fffffffa c16df315 f52ba400 00008001 6e6b6e75 2d6e776f
Call Trace:
 [<c12a8182>] dump_stack+0x16/0x24
 [<c10e2c54>] panic+0x79/0x170
 [<c1847e48>] mount_block_root+0x11c/0x189
 [<c1002930>] ? get_sigframe.isra.0.constprop.0+0x170/0x170
 [<c1848048>] mount_root+0xdb/0xe2
 [<c1848a8b>] ? initrd_load+0x292/0x29c
 [<c1848194>] prepare_namespace+0x145/0x176
 [<c1847b9b>] kernel_init_freeable+0x194/0x1a6
 [<c15f3b3c>] kernel_init+0xd/0xc1
 [<c15f7db2>] ret_from_kernel_thread+0x22/0x40
 [<c15f3b2f>] ? rest_init+0x74/0x74
Kernel Offset: disabled
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

jan-kiszka, 2020-08-25:
Please try https://github.com/siemens/meta-iot2000/commits/jan/queue. It worked for me. Note, though, that this is an intermediate development state. You could try picking 7d008dc9e225d4dd2bbec86c4cf9626cd16b51a8, 5a15f363ce0fdc9f245a16d4f7b42f6daa2b0557 and possibly also eada408945684931998c7415c2e14239f6a86203 if you do not want the full update.

glcj, 2020-08-25:
Hello,
I did the tests and they were successful.

1. I take the "jan / queue" branch and the local repository is updated.
2. Modify the file "wic-image.iot2000.wic" as follows:

#part / --source rootfs --ondisk mmcblk0 --fstype = ext4 --label platform --align 1024
part / --source rootfs --ondisk sda --fstype = ext4 --label platform --align 1024

3. The new configuration is already compiled with kas.

Perform the test with a 4G USB drive and a 250GB SSD drive.

Thankful for the help,

closed 2021-03-02 by jan-kiszka