siemens/meta-iot2000 #141: entropy slow boot

josscheng, 2019-11-08:
Hello

I have a custom image which is quite small and includes the bsp layer. It requires Openssh but is increasing boot times quite a lot even after including haveged. I was wondering if anyone had any advice about how to increase entropy. 

I have tried messing about with kernel fragments by creating a linux-cip_4.4.bbappend and .cfg files etc today but nothing seemed to have an effect.

I know that the example image boot times are quite a lot faster and I was wondering if anything was being done in that layer however nothing obvious stands out other than haveged.

If anyone has any suggestions or advice I would be very grateful

jan-kiszka, 2019-11-08:
Yeah, known limitation of low-end hardware. A while back, I still tried to fight back libraries and applications requesting randomness early and blocking the boot otherwise. Meanwhile (image 2.6), I gave up on that.

I've locally worked around it by using a WIFI stick that has an on-board hw random number generator. But those are rare.

The only generic hope I see: https://lwn.net/Articles/802360/. Didn't look into backporting that kernel change to 4.4 yet, nor did I try of an upstream kernel now shows much better behaviour. There should still be a upstream kernel recipe in the layer, though I didn't used that in a while. If you try it out, please share the result!

josscheng, 2019-11-11:
Thank you for the response. I had already ruled out backporting the kernel for this image for a few reasons, mostly time constraints, but I might look into it as a personal project.

I will move onto looking for a hardware solution for now and if time allows looking into the upstream kernel recipe. If I have any joy with that or the backport I'll be sure to share the results.

jan-kiszka, 2019-11-12:
FWIW, I've backported 14 patches from latest mainline to 4.4 today, including the patch cited above. The result boots much faster, but I'm not yet feeling totally safe regarding the correctness. Will keep you posted.

jan-kiszka, 2019-11-12:
Strangely, I cannot reproduce a slow boot with current master revision anymore. So there is no positive impact of the patches measurable. Rather, it always requires haveged to make /dev/random return bytes. Will retry with just the bsp image.

jan-kiszka, 2019-11-12:
https://github.com/siemens/meta-iot2000/commits/jan/entropy-improvements

josscheng, 2019-11-22:
Thanks a lot for this. It's completely fixed the issue.

closed 2019-11-22 by josscheng